horizon-eda/horizon #363: Renaming power symbol renames all power net

Pitrsek, 2020-03-28:
Version:horizon-2020-03-23-2339
Schematic issue 
Select power symbol, change the name of the power symbol. 
Expected behavior:
Selected power symbol is renamed to new name and connected to power net of new name. If power net of such name does not exist, new power net with such name is created
Current behavior:
Power net to which the edited power symbol was connected is renamed.  all power symbols that were connected to old power net are now connected to a new power net name.

carrotIndustries, 2020-03-28:
>  Select power symbol, change the name of the power symbol.

Have a closer look at the property inspector. You're not changing the power symbol's name, you're changing the name of the net attached to it.

> If power net of such name does not exist, 

Connecting nets by equal name is something I explicitly avoided in the design of Horizon EDA as it can lead to unintended connectivity.

All in all, the behaviour you're seeing is intended and is here to stay.

OTOH, having a tool tool that changes a power symbol's net is something that I actually might implement in the near future.

Pitrsek, 2020-03-28:
You are right, I missed the description in property inspector. I more or less following what I was used to do in other EDA packages.  Having a possibility to change the power symbol's net would be very welcome.  I'd suggest to place the option above the nets part of the properties, to avoid accidentally renaming the net as I did. Since the "rename" is default behavior in the other packages, chances are that I wont be the only one to try do the same in horizon.

closed 2020-03-28 by Pitrsek