horizon-eda/horizon #366: Feature: Waived Rule Violations

iovaris, 2020-04-01:
There are a number of instances where it is preferable to waive a DRC/ERC rather than create a narrowly specific rule to bypass it, or worse disable the rule entirely.

I imagine this would take the form of selecting a rule in the Checks tab, clicking a new "waive" button and providing an optional description/reason. This would result in updating the rule to a yellow "waived" status and being stored in the schematic/board's JSON file for persistence across multiple sessions.

carrotIndustries, 2020-04-01:
The thing is, we currently don't really have a way to track DRC errors across multiple DRC runs which is needed to remember the waived state.

 The best approach I see to fix this is to compute a hash during DRC run based on net, patch types, error polygons, etc.

iovaris, 2020-04-02:
I agree, a computed hash seems like a good approach assuming that it’s easy to maintain when new checks get added. I haven't looked into the code well enough to know.

I'm pretty sure Altium didn't get around to this feature until a few years ago (a 20+ year old program now). KiCad only recently started attempting to implement this (if at all), to my knowledge. Hopefully with the code in Horizon being so lean it wouldn't be too difficult.

carrotIndustries, 2020-04-02:
>  KiCad only recently started attempting to implement this

Can you have look how they did it? Perhaps we could get some inspiration from them.

iovaris, 2020-04-02:
I saw it requested in [this thread ](https://forum.kicad.info/t/suppress-specific-drc-outputs/17164) from May, linking to a user-survey for V6 DRC improvements. Although looking at it now, it may not have made it into their V6 [roadmap](https://gitlab.com/kicad/code/kicad/-/wikis/KiCad-6.0-Roadmap) as far as I can tell.

For inspiration, the Altium wiki outlines this feature quite well: https://www.altium.com/documentation/altium-designer/selectively-waiving-drc-violations-ad?version=17.1
OrCAD's implementation: https://www.orcad.com/cn/resources/library/waive-drc