horizon-eda/horizon #371: Feature Request: Suggested fixes for rule violations

m-byte, 2020-04-06:
While thinking about possible ways of making the footprint creation easier, I was thinking that it would be great if next to some of the rule violations, there would be a bit of text with suggested fixes. Preferrably, this could then include a link to the tool that you shold use. E.g. "no courtyard" would then allow you to quickly get to the "generate courtyard" tool.
Of course, the preferred workflow would be to create the footprint and then only use the rules to check. However, this would help if you thought you were done, run the check and find out something is still missing.

carrotIndustries, 2020-04-06:
Do you see any use for this in the board editor as well?

If not, perhaps a dedicated footprint checklist that updates in real time (as most checks are trivial and fast) with links to the tools to fix the open points might be a better solution.

m-byte, 2020-04-07:
That's a good question... Likely most of the warnings you get there cannot be automatically resolved. For schematic, on the other hand, it would be nice to be given the ability to merge nets with the same name (having worked mostly with Altium and KiCad, assigning nets was something I had to get used to). However, that would probably be best in the context menu.

So overall, you are probably correct in suggesting a footprint checklist. Maybe even like the unplaced section in schematic view.

carrotIndustries, 2020-04-07:
> So overall, you are probably correct in suggesting a footprint checklist. Maybe even like the unplaced section in schematic view.

That was my first thought as well, but this space is already occupied by the layer help. Anyhow, I think that the layer help and checklist could probably be merged as they server similar purposes. 

@atoav What are your thoughts on this?

atoav, 2020-04-07:
@carrotIndustries I like your proposal of having an updating check list that updates automatically

The left sidebar is already a little bit full and it is meant to describe the layers more or less. From a cognitive standpoint it makes sense to seperate the two I think. My proposal is to add something like the heads up display we use elsewhere but with the checklist added:

[image]

Alternatively we could split the lower half of the sidebar into two (vertically) and make the lower half white text on dark, to make it visually distinct from the rest, at the price of having to show a scrollbar (but you get more editor screenspace):

[image]

Both approaches have their pros and cons, but I lean slightly towards the first. What do you think?

carrotIndustries, 2020-04-07:
I lean more towards making the checklist a separate window since that way we have more space for explanatory text and links to tools.

atoav, 2020-04-07:
To be honest I am not so sure about placing it into a seperate window, as this would reduce the discoverability of the whole thing drastically. And discoverability is important here, because this kind of checklist is most valuable to inexperienced users or users who rarely add parts – which means that they would hardly ever be able to make use of this unless they e.g. read it in the manual.

What if the list is there by default, but you could collapse it easily?  (Bonus points for showing a green tick only if everything is okay)  
[image]

That way it is there for beginners and people who create new parts once a year, while power users can hide it and still profit from the small green tick once everything is alright. Dunno how hard it is however to implement such a thing..

carrotIndustries, 2020-04-07:
> To be honest I am not so sure about placing it into a seperate window, as this would reduce the discoverability of the whole thing drastically. And discoverability is important here, because this [...]

Very good point, another idea: Rather than making this checklist HUD style, extend the left sidebar to the right, put the checklist in the top half and the suggestions on how to fix things in the bottom half. Also add a button at the very bottom of the left sidebar that show a summary of the checklist and shows/hides the checklist panel.

atoav, 2020-04-07:
So rather than sliding it out from the bottom HUD style, it would "slide" out from the left as an extra column. That could work as well sure.

I just thought that what I proposed could be a pattern that could be applied to other similar problems as well (e.g. quick displays for ERC, or Board Rules etc.) Achieving solutions that work in similar ways througout the whole software is a value in itself. I especially like this in Blender: every type of pane (be it 3D-Viewer, Timeline, Shader editor, Text editor, etc) has two side panes that can be hidden or displayed using the <kbd>n</kbd> and <kbd>t</kbd> keys or the mouse. One side is for tools and the other for Values and Options:  
[image]  
The cool thing about this is, that once you learned this, you can use that knowledge nearly everywhere and it just works.

Regarding this feature I don't have any real plan with it, but it would be cool if the way it is solved in the UI could be a solution that lends itself to future reuse, beyond this special case here.