horizon-eda/horizon #372: Annotation / Explanation for schematic segments

rroohhh, 2020-04-06:
This is just something I have been thinking about a bit and am not sure about wether its actually useful or if I maybe just haven't found the proper way to do it yet.

For non-obvious parts of a schematic I would like to add explanations like for example the reasoning for a certain circuit or specific calculations to a specific part of a schematic.
Currently I do by just adding text to the schematic, however this text is often quite long / big (compared to the size of rest of the schematic). Also the text tends to include links to things like datasheets, that tend to be long and distracting. In addition to that I wouldn't want to always include these explanations, for example when printing the schematic. 

I am not sure what the best way to improve this is (or if it is actually necessary to improve this), but what I have been thinking about is being able to draw a polygon around a schematic section and then being able to add text that is associated with this polygon, maybe by clicking the polygon and editing the text in the sidebar, or popping up a window to edit the text.
The text would not be visible in the schematic itself, only the polygon. For exporting the polygon would be excluded. Some more advanced text markup atleast for links would also be helpful (maybe similar to markdown?) .

What do you all think?

m-byte, 2020-04-06:
Kind of what Altium is doing? https://www.altium.com/documentation/altium-designer/sch-obj-notenote-ad

rroohhh, 2020-04-06:
Yes that sounds similar to what I am imagining, though I would like to be able to associated a the note with a specifc section of the schematic (like actual components). Maybe Altium can do that, I am not sure just from the documentation and have never used Altium.

carrotIndustries, 2020-04-08:
Apart from the linkage to schematic items, perhaps this could be solved by making the text really small?

If really want to go down the 'sticky note' road, then they sticky note needs to have some distinct advantages over regular texts.

m-byte, 2020-04-09:
I guess it could also be solved with links in the text. Maybe the regex for link detection should be replaced with `((https?|file):\/\/([\w\.-]+)(\/\S+)?)`. Then, one could link to files that contain calculations, explanations, ...

rroohhh, 2020-04-09:
> Apart from the linkage to schematic items, perhaps this could be solved by making the text really small?

Hmm I don't really like that idea, either you end up with text thats unreadably small when printing, or the text is so small, that it is difficult to find it to select it in the editor

> 
> If really want to go down the 'sticky note' road, then they sticky note needs to have some distinct advantages over regular texts.

Well I am not married to the 'sticky note' idea, it would for example also maybe be possible to expand regular texts, maybe with
- a option to collapse the text, that results in it being rendered like the collapsed sticky note in altium
- a option to exclude text from pdf export

fruchti, 2020-04-09:
You can, today, add footnotes to a schematic by adding text ‘Note 1’ to parts of the schematic and including an extra sheet with the explanations. This has the added bonus that no information is lost when exporting a PDF or printing the schematic (which is pretty important for archiving projects IMHO).

I still like the idea of a more convenient way to add background information to portions of a schematic. A few advantages and ideas for sticky notes, in particular, as they are, in my opinion, the most intuitive solution:

* In the simplest case, the note is just represented as a symbol on the schematic. This would allow it to be placed pretty much anywhere while arbitrary amounts of text could fit. When just using regular text, the schematic often has to be ‘planned around’ the text boxes in my experience.
* For PDF export, a sticky note could be turned into a PDF annotation. This preserves most of the functionality and I think PoDoFo supports adding annotations? Alternatively, sticky notes can still be turned into footnotes.
* A sticky note could include a bounding box, which could be visible on mouse-over or when the note is ‘opened’. Thus, even if the collapsed note is just a symbol, it would be unambiguous which components it refers to without extra line drawings cluttering the schematic.
* Multiple types/colours of sticky notes can be useful as well. Most importantly, they can optionally function as a TODO marker Horizon can know about, e.g. making a list of TODOs possible without guesswork about text fields' contents.
* Since the content representation is completely separate from the schematic's, sticky notes could be extended in isolation if they turned out to be really popular. In other words, the limitation to pure text can be softened without introducing problems for schematic text. This is really overkill at this point, but including images (formulas, oscilloscope traces) came to my mind.

So it all boils down to a matter of convenience when adding notes to schematics. So far, I haven't really placed more than a paragraph of text on a schematic but then again, I haven't used any EDA tools having such options so far and my documentation preference is probably shaped by their limitations.