horizon-eda/horizon #378: Symbol and Footprint import

trembel, 2020-04-14:
A nice feature would be to import symbols and footprints from other ECAD software. There are a lot of site providing them like f.e. SnapEda (https://www.snapeda.com) - it would safe a lot time if needed components could be imported somehow.

trembel, 2020-04-14:
and if there's such a thing (I saw a minute ago the issue about importing KiCad footprint), I couldn't find it..

carrotIndustries, 2020-04-14:
The KiCad footprint import is available from the package editor. It's not directly in the pool manager as the import isn't import-and-forget and imported footprints usually require a bit of manual editing to comply with horizon's way of doing things.