horizon-eda/horizon #379: Using components from different pools

trembel, 2020-04-14:
Is there the possiblity to somehow use parts from different pools? I can not find such a functionality, but it would make it possible to create f.e. a pool for myself whilst still using components from the official pool.

carrotIndustries, 2020-04-14:
There are three ways to solve your use case:
 -  Have your own pool include the main pool (settings tab)
 - Use git and maintain your own parts in a separate branch and rebase it onto master as needed
 -  If your own pool doesn't include the main pool (not recommended), create a new pool, include your pool and the main pool and use that pool in your projects

atoav, 2020-05-02:
The way I do it is:
1. I created a fork of the official pool and make sure to add the official pool as a upstream remote (`git remote add upstream https://github.com/horizon-eda/horizon-pool.git`)
2. For new parts I go to the master branch of my fork and create a new branch: `git checkout -b partname`
3. I create the part there
4. I push the branch to my forked remote and click the "Create Pull Request" Button that github convinently shows
5. I write a helpful and understandable Pull Request
6. Back in the terminal I checkout to another branch called work (`git checkout work`), merge my PR there (`git merge partname`) and push it as well (`git push`)

This way I contribute to the official pool while still beeing able to use the parts I made right away. Occasionally I have to fetch the upstream and update my forks master so it is in sync with the upstream (`git fetch upstream` and `git rebase upstream/master`) and then push (you might have to use `--force`)