horizon-eda/horizon #380: KiCad footprint import tool fails

greenscreenflicker, 2020-04-16:
@carrotIndustries , the kicad import tool fails on the following file on the current horizon version.
The file is in the attachment and the postfix needs to be modified.

[image]

[Texas_X2SON-4_1x1mm_P0.65mm.txt](https://github.com/horizon-eda/horizon/files/4488370/Texas_X2SON-4_1x1mm_P0.65mm.txt)

carrotIndustries, 2020-04-16:
Custom padstacks are currently not supported as that'll require creating custom padstacks on the fly. While we could, theoretically, automatically create a new padstack when encountering a custom pad, we'd still end up with more padstacks than needed. This particular package also uses multiple pads on one pad to do mask and paste since KiCad's support for custom pad shapes isn't as nice as ours.

This is a prime example why an automatic one-click importer won't work.

The proper way to solve this would be to ask the user to do the mapping and creation of padstacks on import, but that's easier sad than built.

greenscreenflicker, 2020-04-23:
@carrotIndustries, where is the issue with more pad stacks then required? I do not see any harm. The pad stack could be footprint specific, so it does not pop up in the browser.

Referring to the QFPN20 Footprint, there are also custom pad stacks required.
https://github.com/horizon-eda/horizon-pool/tree/master/packages/manufacturer/stm/ufqfpn20
The importer could just add a subfolder, as fruchti did here.

I understand your motivation for pad stacks: Reducing required work. But it's kind of hard to switch between those two. Just my two cents.

greenscreenflicker, 2020-05-12:
@carrotIndustries: any news?

carrotIndustries, 2020-05-12:
Nope, there are more interesting things to work on right now,