horizon-eda/horizon #382: StupidUser: Tutorial on generating components.

greenscreenflicker, 2020-04-21:
Sorry for this stupid issue. I feel lost in usability, that's why I'm creating this issue and asking for assistance. Maybe you can help me clarifying.
I'm trying to get started with horizon-eda, I want to create the footprint of this component. http://www.ti.com/lit/ds/symlink/tlv742p.pdf
and I'm following this note:
http://www.ti.com/lit/an/scea055/scea055.pdf

So, I obviously, i need to get started by creating a padstack. My first stupid question is, do i need a padstack for every new dimension. Having a look at the pool, it seems to me, that there are a lot of exposed pads. Is this assumption right? I read something about parameterisation, but it looks complicated to me. Maybe a practical, parametized x2son padstack for the outer pads can be created by you?

So, speaking about the X2SON package I would be required to create three padstacks, right?
Where are the padstacks stored correctly in the pool? Ifs there a naming convention?

As far as I see, padstacks must be centered arount zero/origin? Is there a function to center 'em around zero?

Generally speaking, are there any video tutorials on getting started? I feel a little bit lost.

carrotIndustries, 2020-04-21:
Welcome to the dark corners of Horzion EDA :) Yeah, there's currently next-to-none documentation, but users already figured it out: https://github.com/horizon-eda/horizon-pool/tree/master/packages/manufacturer/stm/ufqfpn20

> Having a look at the pool, it seems to me, that there are a lot of exposed pads.
Yeah, these are there necessary since every one of them has a different paste pattern.

Your particular package would require only two custom padstacks as you can use a regular rectangular pad for the center one.

> Where are the padstacks stored correctly in the pool?

Custom padstacks go in a packages' `padstacks` directory. It's the default when selecting a package and clicking 'create padstack'.

> Is there a naming convention?

Not really, use whatever you see fit based on what's there.

Back to the odd-shaped pads: Create a new padstack as described above, then draw the copper shape as a polygon. Then duplicate the polygon, move it to the solder mask layer, and set it's parameter class to 'mask'. Then go to the parameters window, add the solder mask expansion parameter, and type `get-parameter [ solder_mask_expansion ]` into the text field to push the solder mask expansion onto the stack. Next, select one edge/vertex of the solder mask polygon and click 'polygon expand' to insert the appropriate command. Do the same for the solder paste. See https://docs.horizon-eda.org/en/latest/parameter-programs.html for a reference on the language used for parameter programs.

fruchti, 2020-05-01:
> Yeah, there's currently next-to-none documentation, but users already figured it out: https://github.com/horizon-eda/horizon-pool/tree/master/packages/manufacturer/stm/ufqfpn20

To be fair, I also hat a bit of help from you and didn't just get everything right immediately ;).

I was thinking about writing some basic tutorials on part creation and I am starting to have a rough structure in mind. This is, on the one hand, a personal preference because I don't really like video tutorials (it's easier to follow at my own pace in a written tutorial). On the other hand, it should be easier to keep up to date as Horizon evolves.

greenscreenflicker, 2020-05-12:
@fruchti I would love a tutorial. Personally, I prefer video-tutorial, as you see where to click.  It would help me to adopt easier to horizon-eda.

jbayless, 2020-06-27:
Hello carrotIndustries,

Just taking a close look at the linked part, TI's recommendation for this part (eg. page 29) is actually to have the solder mask be smaller than the copper pad. In that case, it would be nice to be able to define the size of the _mask_ to be the nominal size, and then use a parameter program to expand the _copper_ under the mask, as per the datasheet recommendations. Do you know if that's possible?

fruchti, 2020-07-20:
I'm sorry for the lack of updates. I've been unexpectedly busy lately and it's probably for the better if I don't promise a time frame for my tutorial's completion for now. I apologise for getting your hopes up (but I still want to eventually write this tutorial). The only positive aspect is that the tutorial won't be tied to Horizon 1.1, which makes quite a difference with the huge usability and beginner-friendliness improvements we've been getting recently.

I uploaded my initial drafts [here](https://github.com/fruchti/horizon_pool_tutorial). My basic idea was to have a few chapters, each based on one part to have a hands-on experience when following along. Each chapter would introduce different aspects and maybe show a few tricks, as applied to the specific part. The exact workflow would be described in text, interspersed with screenshots and short screen-cap GIFs for maximum clarity.