horizon-eda/horizon #384: Feature request: user-defined parameters

jmgao, 2020-05-04:
I'm creating a footprint for TI's [RKG](http://www.ti.com/lit/ml/mpqf254d/mpqf254d.pdf) package, which has a beveled rectangle for one of its pads, and noticed that there are already two other padstacks that do this.

It'd be nice if there could be a single bevelled rectangle padstack that has `{top,bottom}_{left,right}_{width,height}` parameters that would make it so only one parametric padstack is necessary.

carrotIndustries, 2020-05-04:
Unfortunately, the concept of built-in parameters is more or less set in stone by the `ParameterID` enum: https://github.com/horizon-eda/horizon/blob/master/src/parameter/set.hpp#L9

There are several ways we could get of this:
 - Add a couple of 'custom' parameters and perhaps provide a way of naming them. Feels pretty tacked-on (it is).
 - Use negative/high numbers for custom parameters. Ugly hack as it violates the concept of enums
 - replace `ParameterID` with `std::string` and use the current set of enum items as string constants. As the on-disk format and parameter programs use the strings defined in https://github.com/horizon-eda/horizon/blob/master/src/parameter/set.cpp#L6, this could be done without breaking anything. (If didn't overlook something) comparing strings is a bit slower than comparing `int`s, but I don't think that this will have a significant performance impact.

I initially limited the parameters to a set of built in ones to enforce compatibility, but I can definitely see the usefulness of introducing user-defined parameters while keeping the existing set of predefined ones. To still provide some restriction on the parameter names, I propose adding a lift of legal parameter names to the pool that can be used to validate padstacks and packages.

jmgao, 2020-05-04:
> * replace `ParameterID` with `std::string` and use the current set of enum items as string constants. As the on-disk format and parameter programs use the strings defined in [...]

If we're worried about comparison performance, we can use a tagged union that uses ints for the built in parameters and only falls back to string comparison for custom parameters. Any objection to bumping up to C++17 for `std::variant`?

I'm wondering what the interface for defining custom variables should look like, though. My initial thought is to add something along the lines of `declare-parameter variable-name "Variable Name" [default value]`, but perhaps we might not want to execute all of the parameter programs to figure out what variables are available for validation? Maybe we could just run it at parameter-program modification time, and save details about the parameters in the containing object?

(Or should this be outside of the parameter-program itself, and modified via the UI on the side?)

> To still provide some restriction on the parameter names, I propose adding a lift of legal parameter names to the pool that can be used to validate padstacks and packages.

Sorry, I'm not sure what you mean by this. Do you mean to cache the parameter names in the pool after parsing padstacks/packages from json?

(I'd be happy to implement this, BTW)

fruchti, 2020-05-04:
I'd really like to see custom parameter names! For example I originally wanted to make the bevelled edges of the UFQFPN-20's padstacks parametric to make them more generally usable, but has to skip that, lacking a parameter to adjust it.

Are the existing predefined parameters not sufficient for validation? I don't know if a restriction of custom names to a list would really make things easier. I already made some weird parametric padstacks where I had to use the least wrong parameter name, and even with an extended list such cases would inevitably come up again.

carrotIndustries, 2020-05-04:
> Any objection to bumping up to C++17 for std::variant

Currently, the oldest supported compiler probably is gcc 6.3.0 o debian stretch. As per wikipedia, this version doesn't support C++17 yet. Given that debian stretch will probably reach EOL this year, I think that this is a good opportunity to bump the minimum requirement to debian buster that includes gcc 8.3

> (Or should this be outside of the parameter-program itself, and modified via the UI on the side?)

That's what I had in mind. The menu for adding a parameter would then include a button for adding a custom one.

> Are the existing predefined parameters not sufficient for validation?

Thinking about this for a bit more, they're probably fine as the "contract" about parameter names only exists between packages and padstacks. And people don't expect custom parameter names of one padstack to apply to another.

> (I'd be happy to implement this, BTW)

Cool

jmgao, 2020-05-11:
I started on an implementation of this [here](https://github.com/jmgao/horizon/commits/custom_parameters) that implements everything but the UI to define new custom parameters. I've only really tested it with padstacks, so it's probably not ready to merge, but I thought I'd post this to get some feedback on the approach before poking around further.

Random thoughts that should probably be commented in the code somewhere:
 - I added a ParameterDef type that currently just stores the required parameters and a mapping from id to name: I'm expecting that at some point, someone will want to support custom parameters with different units (rotation, etc.), so we're not just storing the map directly in json.
 - I'm not sure what the policy is on migration of the json format, so I left parameters_required as is, and ParameterDef (de)serializes to/from multiple children of its containing node. If breaking compatibility is okay, then this should probably be moved to a bool field in each parameter.
 - I'm not super happy with [overloading on `[const] ParameterDef*`](https://github.com/jmgao/horizon/blob/634736395fe7f94f495f11061eae36f16896c5d3/src/widgets/parameter_set_editor.cpp#L132) to determine whether or not to show "Add custom parameter" (not implemented yet), but I can't think of a significantly cleaner method.

example with [this padstack](https://github.com/jmgao/horizon-pool/blob/lmz35003/packages/manufacturer/ti/rkg/padstacks/smd-rectangular-bevelled.json):
[image]

carrotIndustries, 2020-05-13:
Hm, not quite sure if the complexity added by special-casing built in parameters is worth the potential performance benefit.

jmgao, 2020-05-26:
I've amended the commit to just use a std::string throughout instead of special-casing the builtin parameters. I took a quick look at implementing the GUI for adding new custom parameters, but got lost. It'd probably be a lot more efficient if someone that knew what they were doing with GTK implemented that instead of me :-)