horizon-eda/horizon #390: Usability feedback

Timmmm, 2020-05-09:
I had a quick play around with Horizon without reading the manual (it's definitely possible to make EDA software that doesn't require a manual for the most part). It's pretty slick - waaaay easier to use than Kicad. Still not as obvious as Designspark PCB which is my current tool of choice, but I thought you might appreciate the feedback anyway:

1. It's not obvious if I use the default pool if I can add my own private components do it, or do I *have* to submit pull requests?
2. On the project "home" screen it would be good to add a "Start here" hint for new users to tell them to click on "Top Schematic".
3. Not obvious how to create a new part. It's one of the most common things to do - should be on the project home screen along with Top Schematic, Board, etc.
4. When placing a big part you can be too zoomed in to see it - it should zoom to the part size if the zoom is too wildly out to see the part.
5. Pool cache table should be resizable.
6. Not obvious why you are put into a "Unit editor" when you create a part. Btw "Unit" and "Entity" are really generic names. Names should convey something about what they are if possible. Failing that I suggest a dismissable line under each tab saying something like "A unit is...".
7. Not obvious what to do in the Unit editor once you've set the name/manufacturer. Would be good if it said something like "No pins, click + to add one".
8. No idea what a "Swap group" is. Tooltip would be good.
9. No idea what to do once you've clicked "Save"? The unit doesn't appear in the list, it seem like nothing happens.
10. Auto map pads does not do what I expect. If I make a generic 20 pin component I'd expect it to assign all pins 1:1, but it doesn't.
11. Can't figure out how to move nets in the schematic editor except via the arrow keys?
12. Can't figure out how to define the board outline at all.
13. Can't figure out how to add freestanding silkscreen text.
14. Gnome's hamburger menu is idiotic (several articles have been written about this). Better to use a normal window menu (File, Edit, etc.). 
15. 3D view actually works! Nicely done!
16. UI is pretty slick.
17. The Main->Pool Manager / Main -> Project system (kind of like Mac System Preferences) is very confusing. I would not expect it to behave like that.

In general this is pretty awesome and I have high hopes for it, but there are still lots of "That button didn't seem to do anything. What do I do now?" moments. Still way better than Kicad though where literally every moment I tried to use it was like that.

Good luck!

Timmmm, 2020-05-09:
I decided to persevere a little, and I read the documentation. While that shouldn't be necessary, it is at least pretty good! More thoughts:

1. Why are "Units" not simply named "Gates"? I can't see a difference between them. Gate is a pretty terrible name too IMO but it is at least standard. Maybe you're trying to draw a distinction between a type of gate, like "NAND" vs "Power Input", and an instance of a gate like "NAND A", "NAND B", but I think it would be less confusing if both of those were called "Gate". You could use "Gate Type" or "Gate Instance" to disambiguate if necessary.
2. It's pretty annoying to have to worry about gates and units when 99.9% of parts only have one.
3. I've *almost* managed to create a part, but I can not figure out at all how to select a symbol for a gate/unit. On the Units page there is "Create Symbol" button but I don't want to create a new one - I am happy with the existing generic ones. There is a Symbol dropdown but it is disabled. I would have expected I can click "Select Symbol" or something, and it would let me choose a symbol. That's how it works in Designspark PCB.
4. It would be good if there was an "Open in File Browser" button in the Pool Manager next to "Update Pool".

Timmmm, 2020-05-09:
For reference, this is how it works in Designspark PCB:

[image]

There's a library manager, which is a separate window. The only types of things are Components, Symbols ("Schematic Symbols") and Footprints ("PCB Symbols"). Each component just maps the pins in a symbol to the pads of a footprint:

[image]

Conceptually it is waaay simpler. I think there are some limitations, like I remember having trouble getting it to realise pins were internally connected - i.e. on a battery connector I don't need to actually connect all three ground pins. And I can't recall exactly how gates work, but honestly I've never had to because it is such a rarely used feature.

Probably too late to change your entire library system, but I think you could definitely improve it by renaming some things, and possibly by adding a "simple part" that is restricted to a single gate ("Unit").

carrotIndustries, 2020-05-09:
Thanks for the perspective from a first-time user.

>  It's not obvious if I use the default pool if I can add my own private components do it, or do I have to submit pull requests?

No, see the replies to #379

> On the project "home" screen it would be good to add a "Start here" hint for new users to tell them to click on "Top Schematic".

Perhaps renaming "Top Schematic" to "Schematic" is good enough since we currently don't have hierarchical schematics.

>  Not obvious how to create a new part. It's one of the most common things to do - should be on the project home screen along with Top Schematic, Board, etc.

From an organisational point of view, this doesn't make sense, that's what the pool manager is for. So how about a button that opens the project's pool in the pool manager?

>    When placing a big part you can be too zoomed in to see it - it should zoom to the part size if the zoom is too wildly out to see the part.

Seems reasonable, I might implement it in the near future.

>    Pool cache table should be resizable.

Which part of the list is too large/small?

>    Not obvious why you are put into a "Unit editor" when you create a part. Btw "Unit" and "Entity" are really generic names. Names should convey something about what they are if possible. Failing that I suggest a dismissable line under each tab saying something like "A unit is...".

Unit is indeed an invention of mine (naming things is hard, loosely inspired by KiCad), but an entity in Horizon EDA is somewhat similar to an entity in VHDL.  Not sure to which extent we should put documentation in the UI.

>    Not obvious what to do in the Unit editor once you've set the name/manufacturer. Would be good if it said something like "No pins, click + to add one".

Done in 8fa9452b467b62227f17a281bbb05a8a9029f74e.

>    No idea what a "Swap group" is. Tooltip would be good.

Good point. Swap groups were there from day 1 to indicate which pins are swappable with one another. (Swap group 0 is not swappable, pins in other groups are swappable with pins of the same group). Even though pin swapping hasn't been implemented so far, I though that it was worth having to have the information there if pin swapping finally lands. Given that pin swapping hasn't made it in the past three years and very few units actually have swappable pins, removing this from the UI is probably the best solution for now if no one objects.

>    No idea what to do once you've clicked "Save"? The unit doesn't appear in the list, it seem like nothing happens.

The pool only gets updated after closing the unit editor. Since the pool update is really fast these days, updating the pool right after saving to reflect the changes seems like something that makes sense to implement.

>    Auto map pads does not do what I expect. If I make a generic 20 pin component I'd expect it to assign all pins 1:1, but it doesn't.

3009711842cf8549195d98dce917cea948b8be91 sets this buttons sensitivity accordingly. That button is intended to be used with connector entities where pin name == pad name. What was your expectation from this button?

>   Can't figure out how to move nets in the schematic editor except via the arrow keys?

Grab and drag the line and start dragging or select it and invoke the move tool by pressing `m`.

>    Can't figure out how to define the board outline at all.

Draw a polygon in the outline layer. Which thing did you look for that didn't work?

>     Can't figure out how to add freestanding silkscreen text.

Place a text on the silkscreen layer. Either press `p t` or search for it in the spacebar menu. In the near future, ther'll be buttons for this: https://github.com/horizon-eda/horizon/issues/119#issuecomment-626062789

>    Gnome's hamburger menu is idiotic (several articles have been written about this). Better to use a normal window menu (File, Edit, etc.).

Nope, not going to happen. Many feature-rich applications have stretched the classical menu bar way beyond its intended complexity and users unfortunately grew accustomed to it. See [that video from microsoft](https://channel9.msdn.com/Events/MIX/MIX08/UX09) on why they moved form the usual file, edit menu to ribbons.

The current way of accessing all features is a searchable menu invoked by pressing space. Discoverability of that menu is currently being worked on. in #119

>    3D view actually works! Nicely done!
>    UI is pretty slick.

Thanks for the nice words :)

> The Main->Pool Manager / Main -> Project system (kind of like Mac System Preferences) is very confusing. I would not expect it to behave like that.

Can you elaborate, what did you expect?

carrotIndustries, 2020-05-09:
> Why are "Units" not simply named "Gates"? I can't see a difference between them. Gate is a pretty terrible name too IMO but it is at least standard. Maybe you're trying to draw a distinction between [...]

See my above comment, naming things is hard and I thinks that there's an advantage to name things unambiguously.

The whole Pool manger UI assumes that the user at least glanced over https://horizon-eda.readthedocs.io/en/latest/pool-why.html

>    It's pretty annoying to have to worry about gates and units when 99.9% of parts only have one.
    I've almost managed to create a part, but I can not figure out at all how to select a symbol for a gate/unit. On the Units page there is "Create Symbol" button but I don't want to create a new one - I am happy with the existing generic ones. There is a Symbol dropdown but it is disabled. I would have expected I can click "Select Symbol" or something, and it would let me choose a symbol. That's how it works in Designspark PCB.

Symbols are directly linked to units. When you want to create a part from an existing entity, then just pick it when clicking create part. Horizon's way of dealing with all of this is certainly more complex than other EDA tools, but allows for easy reuse of items to avoid redundancy.

> Probably too late to change your entire library system, but I think you could definitely improve it by renaming some things, and possibly by adding a "simple part" that is restricted to a single gate ("Unit").

> and possibly by adding a "simple part" that is restricted to a single gate ("Unit").

Go check out the part wizard. (package tab) That one makes creating parts with all-new symbol/unit/entity much more streamlined.

>    It would be good if there was an "Open in File Browser" button in the Pool Manager next to "Update Pool".

Seems a bit to prominent for something that's usually not required. Right-clicking on the pool on the start page will bring up a menu with the desired option.

Timmmm, 2020-05-10:
> Perhaps renaming "Top Schematic" to "Schematic" is good enough since we currently don't have hierarchical schematics.

Yeah that would definitely help. I also think there is plenty of space so you could change the layout to vertical and add some explanatory text. Excuse my MS paint skills:

[image]

> From an organisational point of view, this doesn't make sense, that's what the pool manager is for. So how about a button that opens the project's pool in the pool manager?

Yeah that would be really useful.

> > Pool cache table should be resizable.
> 
> Which part of the list is too large/small?

I'd expect this to be a splitter:

[image]

It seems like it automatically resizes the table to fit its contents, but the fact that the divider isn't a splitter just irks my subconscious GUI convention instinct. I also can't resize the columns. What happens if I have a component with a really long name?
 
> Not sure to which extent we should put documentation in the UI.

Well I think it's definitely possible to go too far, but I have yet to use a tool that did!
 
> Done in [8fa9452](https://github.com/horizon-eda/horizon/commit/8fa9452b467b62227f17a281bbb05a8a9029f74e).

Yeay!

> Given that pin swapping hasn't made it in the past three years and very few units actually have swappable pins, removing this from the UI is probably the best solution for now if no one objects.

Sounds sensible to me!

> [3009711](https://github.com/horizon-eda/horizon/commit/3009711842cf8549195d98dce917cea948b8be91) sets this buttons sensitivity accordingly. That button is intended to be used with connector entities where pin name == pad name. What was your expectation from this button?

Ah yeah I eventually figured out that I had to select all the pads on the right and *then* click "Auto map". Since it was enabled with none selected I figured it should do something without having to select any pins (hopefully map all of them!). I think if it is disabled until you select something that might be good enough.

It also wasn't terribly obvious to me that I *could* select multiple pads - you can't select multiple pins and there's no visual difference between their UIs.

> > Can't figure out how to move nets in the schematic editor except via the arrow keys?
> 
> Grab and drag the line and start dragging or select it and invoke the move tool by pressing `m`.

Ah yeah I found that in the documentation. Maybe a few common shortcuts could be added here:

[image]

Or you could make the black tool row (which is great btw) always present:

[image]

Incidentally it took me quite a while to realise that track routing is modal, and that you have to press enter (or escape? hard to know which one!) to exit it. Also it would be nice if undo worked within the tool.

I think a traditional photoshop-style tool box would really help here.

> > Can't figure out how to define the board outline at all.
> 
> Draw a polygon in the outline layer. Which thing did you look for that didn't work?

Ah yeah, found that out from the documentation too. I dunno really, I guess some kind of "Define board outline" tool? Again a toolbox would help here. Also, in some part of the program (I think the package editor?) you have a nice help text shown under the layer selector that explains what they are. That is *awesome*. I hadn't come across "courtyard" before. It would be nice if it did the same thing in the board editor.

> > ```
> > Can't figure out how to add freestanding silkscreen text.
> > ```
> 
> Place a text on the silkscreen layer. Either press `p t` or search for it in the spacebar menu. In the near future, ther'll be buttons for this: [#119 (comment)](https://github.com/horizon-eda/horizon/issues/119#issuecomment-626062789)

Ah cool, once I found the spacebar menu everything was a lot easier!

> Nope, not going to happen. Many feature-rich applications have stretched the classical menu bar way beyond its intended complexity and users unfortunately grew accustomed to it. See [that video from [...]

Sure but the hamburger is exactly the same thing except now you only get one unnamed menu which is even worse! I think ribbons are a reasonable interface though. Does GTK support that?
 
> > The Main->Pool Manager / Main -> Project system (kind of like Mac System Preferences) is very confusing. I would not expect it to behave like that.
> 
> Can you elaborate, what did you expect?

I guess the weirdness is that it treats projects and the pool manager as the same kind of thing. In the OSX preferences it kind of makes sense because the keyboard settings and mouse settings *are* the same kind of thing, and if I'm changing mouse settings I'm probably not changing keyboard settings at the same time. But in Horizon I definitely want to have a project open and also edit the pool, but it doesn't seem like I can do that.

> Thanks for the perspective from a first-time user.

Thanks for the reply! Since my last message I went back to Designspark PCB but it couldn't do something I wanted - obround through holes. So I went back to Horizon and managed to figure out the difficult buts and have now ordered a PCB. Definitely will be sticking with Horizon now.

[image]

It's a very simple board for injecting messages into the HDMI DDC/CI channel. (Feel free to tell me how many horrible mistakes I've made!)

[... cut ...]
I'd highly recommend using Azure Pipelines for that. The interface is pretty slick, and they provide free builders for Windows, Linux *and* Mac. I've used other CI systems like Travis and Appveyor, and they just aren't as good. Travis has a pretty terrible UI, doesn't support Windows and Mac and Linux are always horribly outdated. Appveyor is much better but it doesn't support Mac.

It's not open source but its so much better than the competition I don't think it's sane to pick anything else. I could set it up for you if you like? Here's [one I did earlier to build MyPaint on Mac](https://github.com/timmmm/mypaint_all) (I gave up because GTK doesn't support pressure on Mac).

Timmmm, 2020-05-10:
> Symbols are directly linked to units

Does this mean that the relationship between units and symbols is 1:1? You can't have two units using the same symbol? If so what is the point of having separate files for them?

carrotIndustries, 2020-05-10:
> Does this mean that the relationship between units and symbols is 1:1? You can't have two units using the same symbol? If so what is the point of having separate files for them?

Why would you want two units using the same symbol? In Horizon EDA, Units+Entities are the single source of truth for the netlist representation of a part contrary to (most) other tools where the symbol defines the pins. A symbol merely is a graphical representation of a Unit.

Symbols are separate from units for two reasons:

 - Separation of netlist and schematic
 - Have two symbols for one unit (think EU and US symbol for resistor)

> so you could change the layout to vertical and add some explanatory text. Excuse my MS paint skills:

Hm, not too bad, @atoav your thoughts?

> >    So how about a button that opens the project's pool in the pool manager?
>
> Yeah that would be really useful.

Done in 66e66c6f1fe7dd349155813114dc96c705c1732c 

> It seems like it automatically resizes the table to fit its contents, but the fact that the divider isn't a splitter just irks my subconscious GUI convention instinct. I also can't resize the columns. What happens if I have a component with a really long name?

The name column will get bigger, but I get your point, so we now have a movable splitter d093adf87cf8b7a611bbcde5f8c1b78657888e28

> It also wasn't terribly obvious to me that I could select multiple pads - you can't select multiple pins and there's no visual difference between their UIs.

I see, do you have any ideas to improve on this? I'm also n

> I think ribbons are a reasonable interface though. Does GTK support that?

Not really and I try to stick to the GNOME interface guidelines as best as possible. and I currently don't see the use case for ribbons. The hamburger menu is only used for things that have global scope and there aren't too many of them.

> But in Horizon I definitely want to have a project open and also edit the pool, but it doesn't seem like I can do that.

You can. There's an option "New window" in the hamburger menu or you just open another window by starting the app a second time. Pool and project manager started out as separate apps but got eventually merged since having two entry points for the app was deemed as too confusing.

> Feel free to tell me how many horrible mistakes I've made!

Depends on what resolution you want to run with this adapter, but HDMI usually runs at several Gbit/s and thus requires differential pairs with controlled impedance.

> one area that looks like it needs improvement is CI - there's only a Windows download at the moment and not for the latest release.

There is, 1.1.1 is bugfix release that just fixes the `install` target in the Makefile, so there's not difference for the users compared to 1.1.0. I recently moved to github actions for CI and don't have much motivation to move to something else. Mac OS support isn't going to happen in the near future as there are some severe bugs/incompatibilites in Gtk's OpenGL implementation on mac os, so providing mac os builds won't do any good.

Which other downloads do you wish for? Currently, Horizon EDA is distributed in these ways:

 - Source tarball for building it yourself
 - Windows MSI installer
 - Debian package (you want to use the one from sid)
 - AUR package
 - Flatpak

I recently had a look at building a PPA to provide an easy-to-use way for ubuntu users, but the process of building a PPA was too complicated for me to bother.

> Incidentally it took me quite a while to realise that track routing is modal, and that you have to press enter (or escape? hard to know which one!) to exit it.

All tools are modal. Both the draw net line and route track tool stay active until you explicity exit them by either right click or escape.

> Also it would be nice if undo worked within the tool.

Unfortunately, that's not going to happen, as undo/redo is handled outside of tools.

> Ah yeah, found that out from the documentation too. I dunno really, I guess some kind of "Define board outline" tool?

In every PCB layout tool, I've used so far the board outline is defined by drawing something (either line or polygon) in the appropriate layer, so horizon EDA isn't special in that regard. But I see that a draw outline tool that's just the draw polygon tool with the layer set to outline could make things more discoverable. @atoav what do you think of this?

atoav, 2020-05-10:
> Hm, not too bad, @atoav your thoughts?

I am all for it. Maybe we can find a more concise wording to avoid users glazing off into the distance and ignoring a wall of text, but this is definitly more helpful than nothing.

> >   Also it would be nice if undo worked within the tool.
> Unfortunately, that's not going to happen, as undo/redo is handled outside of tools.

I understand if you don't want to do this, but I had to think about the way Photoshop does this in it's Bezier Path tool: when you draw a path there, you can press <kbd>Backspace</kbd> and it will delete the last point you created without exiting the tool. This could be useful for the `Draw Polygon` and the `Route Track` Actions alike, as it allows users to undo accidental actions without exiting the tool.

> But I see that a draw outline tool that's just the draw polygon tool with the layer set to outline could make things more discoverable. @atoav what do you think of this?
Why don't we treat this a little bit like the Footprint generators for the Package editor? So some kind of wizard to generate different PCB shapes one would commonly use. So stuff like:
- Simple rectangular PCB (width x height, corner radius)
- Rectangular PCB with mounting holes
- Standard messurements like PCBs for stuff that has a 3U frontpanel etc

etc. whatever you can think of. Could be really practical

Timmmm, 2020-05-10:
> I see, do you have any ideas to improve on this [the pin/pad mapping tables]?

Hmmm it's definitely tricky! How about a "Select all" button above the pads, but not above the pins? Not sure about that though. Designspark PCB does it with a single table which I feel is a little more understandable, but I also found it really convenient that you can map the pads one by one by double clicking on the pads (and it automatically advances - nice work)!

Perhaps the tables should be swapped, so it's like "here are the things you can change on the left, and here are the values it can have on the right", which maybe feels more normal? Not sure about this though.

> HDMI usually runs at several Gbit/s and thus requires differential pairs with controlled impedance.

Ha yeah I figured that might be an issue but hopefully it will work at some low resolution for testing purposes. I have no idea how to route differential pairs anyway so 🤷 

> Mac OS support isn't going to happen in the near future as there are some severe bugs/incompatibilites in Gtk's OpenGL implementation on mac os, so providing mac os builds won't do any good.

Ah ok that is a shame. Might be worth noting in the Readme or Releases.

> Unfortunately, that's not going to happen, as undo/redo is handled outside of tools.

Ah shame. I can live without it though. It took Blender like 10 years to implement it so you've got time!

> Maybe we can find a more concise wording to avoid users glazing off into the distance and ignoring a wall of text.

Yeah I definitely made it a bit long. :-)

Thanks for the great software, and for already fixing loads of these issues - I am really impressed!

carrotIndustries, 2020-05-11:
e9305b327022187b9ad63c928eeb7911a439e4a3 triggers the pool update as soon as something's saved. The old behaviour was a holdover from the days when we didn't have fast incremental pool updates.

0119cb8782c998464d43bae34675a0756f4d3c20 and the previous commit restructured the entity editor: I hope the help buttons aren't too distracting.

[image]

atoav, 2020-05-11:
Both very welcome changes : )

Timmmm, 2020-05-11:
Excellent work!