horizon-eda/horizon #391: Degenerate polygons handled incorrectly

jmgao, 2020-05-11:
The following degenerate polygon renders as a triangle instead of a square:
```
get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] 2 /

set-polygon-vertices [ copper_polygon 5 ],
```
[image]
[degenerate.json.txt](https://github.com/horizon-eda/horizon/files/4607399/degenerate.json.txt)

carrotIndustries, 2020-05-11:
Everything's behaving as expected: You put 6 coordinates on the stack, but only use 5, so the first one 
```
get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /
```
is unused. Replacing the last line with `set-polygon-vertices [ copper_polygon 6 ]`, everything behaves as expected.

jmgao, 2020-05-11:
Sorry, I wasn't paying enough attention when trying to minimize a test case, and this looked the same.
The real problem I was running into appears to be ending on the start point of the path:

```
get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

set-polygon-vertices [ copper_polygon 5 ]
```

m-byte, 2020-05-13:
But that's also a triangle since you only give it three different points: [-2, -2] [2, -2] [2, 2]
The last two points are identical to the first one.

jmgao, 2020-05-26:
Ugh, I should stop trying to reduce this.

Here's [the actual problem I'm having](https://github.com/jmgao/horizon-pool/blob/lmz35003/packages/manufacturer/ti/rkg/padstacks/smd-rectangular-bevelled.json). It's an octagon, but with the corner offsets set to zero, it's a hollow square instead of a solid square:

```
get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

set-polygon-vertices [ copper_polygon 8 ]
```

[image]

Deleting the last point results in this:
```
get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] -2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] 2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] 2 /

get-parameter [ pad_width ] -2 /
get-parameter [ pad_height ] 2 /

set-polygon-vertices [ copper_polygon 7 ]
```
[image]

m-byte, 2020-05-27:
To be honest, I wouldn't necessarily consider this incorrect handling since you are essentially creating a cutout by repeating the points in reverse order. Of course, the remaining polygon is essentially a 0 width line. Once you remove one point, you are only subtracting a triangle that overlaps one corner of the rectangle. So I would expect this behavior.