horizon-eda/horizon #393: Scrolling can accidentally change values in Unit and Entity editors

Timmmm, 2020-05-11:
Thought I'd better start making separate issues rather than adding to #390!

If you scroll with the mouse wheel and while scrolling the mouse happens to land on top of a dropdown or spinner control, then it will stop scrolling and instead change the control value. Took me a while to figure out how some of the swap groups had been changed to 1 or 2. Also an issue in the Unit Editor. In this gif I'm only using the scroll wheel:

[image]

carrotIndustries, 2020-05-11:
Well, this is bit unfortunate since gtk works that way: https://gitlab.gnome.org/GNOME/gtk/-/issues/593

The best we could do is disable scrolling for the dropdowns and spinners.

Timmmm, 2020-05-11:
> The best we could do is disable scrolling for the dropdowns and spinners.

Yeah I think that is the most reasonable workaround.

closed 2020-05-11 by carrotIndustries

Timmmm, 2020-05-11:
Nice one!

commit 6e15a9e0 (2020-05-11): unit/entity editor: remove scroll events from widgets in scrolled window

fixes #393