horizon-eda/horizon #394: Feature request: make all tables sortable by their columns

rroohhh, 2020-05-12:
I often find myself clicking on the table column headers with the expection that the table would get sorted by the column I clicked on, this is however not the case for most (all?) tables present in horizon.

carrotIndustries, 2020-05-12:
It depends on where you look. In pool browsers, all columns that make sense to sort are sortable.

The two examples you mentioned are a bit special as the sorting is done by the BOM/PnP export logic rather than by the treeview itself and I haven't yet bothered to hook up the treeview sort stuff to the export settings.

rroohhh, 2020-05-13:
Ah makes sense, I have been using the BOM export window, to get an overview over the components rather than actually exporting a BOM, so I don't really care about the export order, but of course decoupling the order of the treeview from the export order would make things more confusing.

rroohhh, 2020-05-19:
Another place I just found is the many tables in the pool manager. Most interesting being probably sorting the packages after their number of pads.

carrotIndustries, 2020-05-19:
e53098d481488b49ca6912fb4b40b46d04aebae5 adds sorting by pads