horizon-eda/horizon #399: Import a graphic / image / logo

mattjudge, 2020-05-17:
It would be great to have the option of importing a graphic / image / logo as e.g. a png or bitmap directly into one of the layers. Apologies if this has previously been discussed, or is already possible - I'm a new user, so may well have missed it!

carrotIndustries, 2020-05-17:
Bitmap images don't work well with gerber apart from importing every pixel as a polygon or other ugly hacks, so that's a no.

The preferred way to get artwork into horizon is as such:

 - vectorize in inkscape
 - import dxf
 - line loop to polygon if the shape is supposed to be filled
 - scale to taste

mattjudge, 2020-05-18:
Okay, thanks. I guess from a user perspective it would be quite handy to have the option, but I totally understand why that would not be an ideal solution - I'll close the issue.

Interested to see where this project heads, it'll be great to have another player on the scene!

closed 2020-05-18 by mattjudge