horizon-eda/horizon #400: Minor nitpick: update current selection according to selection filter

rroohhh, 2020-05-21:
I am not sure if this is what everybody would expect, but I would expect that if the currently selected entity is no longer part of the selectable according to the selection filter, it should be deselected.

carrotIndustries, 2020-05-21:
Implementing this is pretty trivial, but should items get re-selected when re-enabling the respective checkbox the selection filter? This makes things significantly more complex since we'd need to remember the selection prior to filtering.

rroohhh, 2020-05-21:
The way I envision it is, that the selected entity would only get deselected after closing the selection filter window. 
With that I don't think its necessary or intuitive to re-select when re-enabling the respective checkbox.

fruchti, 2020-05-21:
IMHO, this could also be a separate tool, which is only available when something is selected already (‘Edit selection’?). The window would look similar to the selection filter, but have ‘deselect’ and ‘only this’ buttons instead of the checkboxes. Possible advantages:

* It is immediately clear that activating a checkbox again won't ‘bring back’ previously selected items.
* At least for me, it's pretty common that I only need the selection filter for one selection, e.g. when I want to select all silkscreen graphics. If the edit selection tool would not modify the selection filter, this would be easier because I wouldn't have to manually reset the selection filter after using it.

It would be possible to disallow the selection filter when there already is a selection, because the use case where you select something, set the filter and then add/subtract to that selection, is probably negligible. When both tools are mutually exclusive, they could have the same hotkey and thus the behaviour @rroohhh envisioned.

rroohhh, 2020-05-21:
@fruchti would the 'Edit selection' tool also edit the selection filter?
Because my usecase is mainly, that I try to select something, but something different I didn't want gets selected (for example, trying to select a via under a bga package and the bga package gets selected instead).
I then proceed to change the selection filter, but the thing I didn't want to get selected is still selected, so I have to manually deselect it.

fruchti, 2020-05-21:
> @fruchti would the 'Edit selection' tool also edit the selection filter?

I think it shouldn't, because most of the time I need the selection filter only once, e.g. to select some artwork on a silkscreen layer to move or resize it. ‘Edit selection’ would replace the selection filter there (select, then drop everything else from the selection just once), while the selection filter would still be useful when I'd have to do something similar multiple times.

rroohhh, 2020-05-21:
I see, then it doesn't really help with my usecase.

fruchti, 2020-05-21:
Ah, then I misunderstood you, sorry. I thought you were also just concerned with the currently active selection and not the next ones you'd make.

carrotIndustries, 2020-05-21:
> The way I envision it is, that the selected entity would only get deselected after closing the selection filter window.

Seems a bit too surprising IMO.