horizon-eda/horizon #402: RFC: Collaboration mode

m-byte, 2020-05-27:
For more complicated boards, it would be nice to have the ability to simulataneously edit schematic or layout with multiple users. I'm imagining something similar to what is described as a feature of Xpedition: https://www.mentor.com/pcb/xpedition/team-layout/

Right now, I have a simplistic proof of concept of this for the schematic editor, but it is a bit rough around the edges (will try to push the branch to my repo later this week):
I am using gloox to connect to some XMPP server and then join a multi user chat.
Whenever `history_push()` is called in `core_schematic.cpp`, a json diff of both the block and schematic is generated. This is then packaged into an extended message stanza (message so that the server will keep the log) and sent to the multi user chat.
Whenever a custom message stanza is received, the diff is used to create a new `HistoryItem`.

I would really like to get some comments on the following:
1. Should there be a separate thread for this when collaboration is enabled in the settings? If so, should this reuse the broadcast socket? Or should this be part of the main_window with e.g. the broadcast socket for communicating with schematic and board editor?
2. Should the name of the chat room be configurable per project or should they be part of the more general settings?
3. Should chat be also available on the UI?
4. How should this be integrated with history? I believe the most user friendly way would be to only use undo/redo on ones own changes. However, there might be cases where changes made by someone else are supposed to be undone. Possibly, this could be part of a chat/collaboration UI? Also, how should undo/redo be propagated to other users? A special message with each diff being assigned a UUID?
4. Are there any objections to merging this once it is somewhat polished? (Probably should have started with this)

carrotIndustries, 2020-05-28:
I think the answer is 'not yet'. Getting this to work so that it meets expectations set by other real-time collaboration tools such as etherpad or office 356 will probably be quite a challenge. Some things that came to my mind:

 - suppose a message about a change arrives while the user is still in a tool and applying the diff as tool finishes doesn't work due to merge conflicts. What to do in this case?
 - How to deal with spotty network connections?
 - Rules and export settings are edited outside of the history
 - diffing the serialized json is a bit of a hack: ideally, we'd need to know what a tool did when it's done. Right now, it's free for all. That'd also help with faster re-rendering in tools and faster rebuilds.

I haven't used expeditions team layout feature so far, but the schematic collaboration is built around exclusively locking schematic sheets to avoid having to deal with merge conflicts. That approach isn't workable for us since sheets aren't self-contained.

m-byte, 2020-05-29:
> * diffing the serialized json is a bit of a hack: ideally, we'd need to know what a tool did when it's done. Right now, it's free for all. That'd also help with faster re-rendering in tools and faster rebuilds.
Yes, I agree that diffing the serialized json is a hack. Ideally, the history would be refactored to have any changes go through some get/set function. Then it would be easy to have the history as a series of transformations.
TP1 being a lot easier than TP2 (https://en.wikipedia.org/wiki/Operational_transformation#Transformation_properties) is why I would use a centralized server to keep track of the order of changes. Here, XMPP is an easy solution, as there are many mature server applications available.

> * suppose a message about a change arrives while the user is still in a tool and applying the diff as tool finishes doesn't work due to merge conflicts. What to do in this case?
With a refactored history, the change would be applied while the user is editing. Especially in the beginning, this could also be counteracted by a) users being at least a little mindful of what others are doing and making sure not to interfere or b) "locking" parts of the board / schematic close to the cursor. Locking could be done as part of the refactored history.

> * How to deal with spotty network connections?
The client would keep track of the changes and send the individual transformations to the server once it has a connection again. It would also load the transformations on the server and apply them before the just uploaded transformations to sychronize with the server state.

> * Rules and export settings are edited outside of the history
I don't necessarily see this as an issue. In general, synchronisation of rules should be added. Since it is possible to split the transformations into mutliple "classes" (schematic, block, board, rules, ...), synchronisation of these could also be added.

> I haven't used expeditions team layout feature so far, but the schematic collaboration is built around exclusively locking schematic sheets to avoid having to deal with merge conflicts. That approach isn't workable for us since sheets aren't self-contained.
Neither have I, but I also understand it this way. For schematic entry, "locking" a sheet would probably mean preventing write access to the node inside the `sheets` object. Also, I believe that the schematic is more difficult to get right for this, so locking sheets is a relatively low effort solution to preventing merge conflicts. I believe locks should be implemented as a list of <UUID, JID> pairs. Whenever write access to an item with a UUID from this list or any child of it is requested, this would be denied unless the JID matches ones own. The XMPP MUC history would keep track of any (un)locking operations. This way, (un)locking would only be possible with a server connection meaning that double locks are prevented. Also, locking would not necessarily be limited to sheets, but could also be applied to e.g. parts on a board.