horizon-eda/horizon #404: FreeBSD build is broken

carrotIndustries, 2020-05-28:
@endofexclusive Seems like the version of podofo that ships with freebsd is incompatible with C++17(?): https://cirrus-ci.com/task/6216188764094464?command=main#L540 any idea how to get the build into working order?

endofexclusive, 2020-05-28:
The LLVM `libc++` has a macro to enable the std::auto_ptr.
https://libcxx.llvm.org/docs/UsingLibcxx.html#c-17-specific-configuration-macros

A possible workaround could be:
```
-D_LIBCPP_ENABLE_CXX17_REMOVED_AUTO_PTR
```
It was mentioned as a workround on the `PoDoFo` mailing list:
https://sourceforge.net/p/podofo/mailman/message/36486453/

I am no longer on FreeBSD so I can unfortunately not try this out.

Interestingly, on Debian 10, the `export_pdf.o` compiles just fine with clang 8.0.1, even though the same `std::auto_ptr` is in the header. So that combination seems less strict.

carrotIndustries, 2020-05-28:
Thanks for the suggestion, the makefile now passes that option if we're on freebsd. Podofo upstream not having resolved this yet makes me wonder if podofo was that good a choice after all...

commit 136157e0 (2020-05-28): fix freebsd ci

closes #404
closed 2020-05-28 by carrotIndustries