horizon-eda/horizon #406: Board editor redraw rate and lines changing shape when panning.

bob-u, 2020-06-07:
I love how smooth and fast zoom in out is.  However, when panning, redraw rate is jumpy (much slower than dragging a window, for example) - is there a way to fix it? 

Also, lines seem do 'dance': lines change thickness and brightness, circles have crawling edges.  I tried to capture it here:

https://streamable.com/9278e3

https://streamable.com/81swki

 Is this a fault of Intel integrated GFX (UHD Graphics 620)?

Lastly I cannot seem to find a way to change the color of the cursor.  Beyond changing color is there a way to have similar functions to a mouse cursor to increase visibility (XOR, dual color, etc).

carrotIndustries, 2020-06-07:
> Lastly I cannot seem to find a way to change the color of the cursor. 

You can change it in the preferences: 
[image]

The cursor is in fact dual color (see the black border):
[image]

Unfortunately, no one has yet figured out how to profile the OpenGL rendering in horizon to find out where its speed could be improved, so there's not much we can do about this for now apart from disabling MSAA (if enabled).

The circle's pixels changing is dependent on how the coordinates line up with the pixel grid. Changing the minium line width to 2px might improve things.

carrotIndustries, 2020-06-14:
017049d57ad5c8f081c77166011cff2bc2fab412 made the grid appearance more consistent

bob-u, 2020-06-18:
I just installed windows and tried v1.1.0 and windows version does not exhibit the issues that I had seen in Linux.  Circle outlines are not fluctuating (crawling) or lines are not getting thicker and thinner on windows like they do on Linux when panning.