horizon-eda/horizon #408: Net color in board editor

rroohhh, 2020-06-08:
Sometimes it would be nice to assign different net classes different colors to make it easier to see and follow them. (For example to keep track of the different groups in DDR memory layout).

This color could for example be displayed by colored dots on the actual track. (Similar to how altium does it)

bob-u, 2020-06-08:
That is a very nice feature to have.  Different color not only for airwires but also for traces themselves.

carrotIndustries, 2020-06-08:
> This color could for example be displayed by colored dots on the actual track. (Similar to how altium does it)

Do you have link that shows how this looks like? From an implementation perspective, recoloring the whole track would be easier of course.

How many colors should we make available for this mode? All of them would then be user-definable in the preferences dialog.

rroohhh, 2020-06-08:
> Do you have link that shows how this looks like? From an implementation perspective, recoloring the whole track would be easier of course.

[Here](https://www.altium.com/documentation/sites/default/files/wiki_attachments/297389/NetOverrideColor_AD18_0.png) is a example of how it looks in Altium. Actually Altium has a lot of different options on how this is visualized (see [this](https://www.altium.com/documentation/sites/default/files/wiki_attachments/297389/ColorOverrideSettings_Examples_AD18_0.png)).

Just recoloring the whole track would be fine as a start aswell for me. 

> 
> How many colors should we make available for this mode? All of them would then be user-definable in the preferences dialog.

I don't think it makes sense to limit this to a certain number of different colors. To me it would make sense to have a interface similar to the rules interface, where you have multiple rules that match on {net class, net, net name [regex]} that allow to specify a arbitrary color.

bob-u, 2020-06-08:
Net color selector in EasyEDA:
[](https://docs.easyeda.com/en/PCB/PCB-Net/)

Green is ground, orange is +3.3V, blue is SPI Clock:
[](https://streamable.com/el9ogj)

carrotIndustries, 2020-06-08:
The thing is, the OpenGL renderer is fairly constrained in which colors can be put on screen: https://github.com/horizon-eda/horizon/blob/master/src/canvas/triangle.hpp#L45 is the definition of a vertex that gets sent to the GPU.

To save space in the vertex data, it effectively uses an 8bit indexed color palette with 0 being the color of the layer. The palette colors get uploaded to the GPU using a uniform buffer object.

It all boils down to whether we quickly want something that works okay-ish or can wait a lot longer for a better solution.

The quick and easy way would be this: Add n (10?) colors to the `ColorP` enum. Provide a way of telling the various `render()` methods in the canvas to use an override color based on net.

The main disadvantage I see in this approach is that there's a only fixed set of colors to pick from and the information which layer a track/pad is on is visually lost.

On the long run, this is related to #313 as that issue also calls for more display styles.

rroohhh, 2020-06-08:
Ah I see, I have no problem with something that works okay-ish first. Maybe we even come to the conclusion that the more sophisticated isn't even necessary.

Working within the constraints of the current renderer, a hacky way to do the checkerboard could be to add a checkerboard type to the `Type` enum and drawing everyhing twice, once with the base color and normal rendering and once using the checkerboard type with the selected color.

My main worry with adding a fixed number of colors is (besides running out obviously) cluttering up the preferences dialog. But I guess ~10 more entries won't be too bad.

Besides 10 colors would work for the things I am thinking of.

bob-u, 2020-06-09:
@carrotIndustries Since you mention long term solution, are you considering moving away from OpenGL since it's deprecated on MacOS, some library based on Vulkan API perhaps?

carrotIndustries, 2020-06-09:
Nope, OpenGL in horizon is here to stay. I like to stick to 'too big to fail' APIs such as OpenGL rather than some random vulkan library. Vulkan itself is a bit too close to the metal to my taste.

Apart from that the limitations aren't with OpenGL, they're 100% with me learning as I go along.

> , a hacky way to do the checkerboard could be to add a checkerboard type to the

Then current OpenGL rendering implementation is already full of hacks and oddities. It's probably about time to clean it up:

Initially (well not initially, but good enough for now) we've had `vector`s of `Triangle`s that got loaded into one buffer and rendered in a single draw call per layer. This meant that the shader had to figure out what to draw based on the vertex data. That shader was rather complicated and probably not very fast: https://github.com/horizon-eda/horizon/blob/a5ad666c4344fe360c349c3f14accca92a0f13f9/src/canvas/shaders/triangle-geometry.glsl

To improve on this, the vertex data is now inspected beforehand and an element buffer is filled with the indices of the vertices ordered by shader. So we load a shader and then draw a range of elements that point to the vertices that are supposed to be rendered with that particular shader.

My plan for improving the renderer takes this concept a step further:
We'll split the vertex data into two parts by means of "parallel `vector`s". One for the GPU, one to provide additional information for sorting the vertices and assigning them to the correct shader. That way we can increase the amount of extra data we can store per vertex such as color or fill patterns and group the vertices such that the shaders need to to even less.

carrotIndustries, 2020-06-12:
The recent commits to master significantly cleaned up  the OpenGL rendering.

What do you have in mind in terms of UI? Having a list of nets with the option to set the color for each one or something similar to rules?

bob-u, 2020-06-12:
List of nets and corresponding colors would be fine, however, sometimes I wished there was a way to use a pattern matching to for example color all nets with 3V, or CLK in them.  

Thinking about this more, it would be good for color to change somewhat when switching layers or perhaps - instead of unique color - use a pattern so color changes according to layer color while pattern is applied to specific net.  Patterns however are not as easy to distinguished as color, especially at alpha <75% and it would require for layers not to use a pattern.

rroohhh, 2020-06-13:
> The recent commits to master significantly cleaned up the OpenGL rendering.

Very nice!
> What do you have in mind in terms of UI? Having a list of nets with the option to set the color for each one or something similar to rules?

Something similar to rules would be very useful I think.

carrotIndustries, 2020-06-13:
Thinking about it a bit more it probably makes sense to merge the UI for this with the airwire filter and just call the window "Nets":

 - Both show a list of nets that the user should be able to search and filter
 - Having looked at the WIP axiom rev3 micro board coloring airwires can probably help a lot in getting a sense of which net goes where

rroohhh, 2020-06-13:
Yeah, that would work aswell and could also improve the airwire filter in the same step, for example to support filtering based on regex.

carrotIndustries, 2020-06-21:
e4842cae282bf63f514b199e848397dcfedb61c1 enhances the airwire window

bob-u, 2020-06-22:
I think Airwire window should have a search feature, like in components.  For more visual input, I wonder if schematic to board object highlighting could be extended as interface for net filter/highlighting.

carrotIndustries, 2020-06-22:
> I think Airwire window should have a search feature

it has one since the commit mentioned above:
[image]

rroohhh, 2020-06-22:
> [e4842ca](https://github.com/horizon-eda/horizon/commit/e4842cae282bf63f514b199e848397dcfedb61c1) enhances the airwire window

Nice, I like the nets window / improved airwire window so far!

closed 2020-06-26 by carrotIndustries

bob-u, 2020-06-26:
Looks great.  Interesting effect along with shadow effect.
One minor problem is that airwire does not get colorized once custom color is assigned- only trace does.  Airwire will get assigned color if checked/unchecked or when trace is modified.  It seems as if airwires are not getting redrawn after color selection is made.

rroohhh, 2020-06-26:
@carrotIndustries lmk if I should create seperate issues for these.

First, nice work, this already helps a lot. Some possible improvements would include:
- Making the settings made in the nets window persistent
- Add a way to have the net color be a checkerboard (or similar) pattern, that allows to see the layer color (currently its hard to see which layer a track is on when using net colors)
- The coloring breaks when moving tracks (using `m`)

bob-u, 2020-06-26:
> Add a way to have the net color be a checkerboard (or similar) pattern, that allows to see the layer color (currently its hard to see which layer a track is on when using net colors)

Custom patterns would be nice to have.  For right now, I found that if you apply pattern to a layer - custom net color will inherit the pattern from the layer it's on so you can see which layer it's on.

rroohhh, 2020-06-26:
> Custom patterns would be nice to have. For right now, I found that if you apply pattern to a layer - custom net color will inherit the pattern from the layer it's on so you can see which layer it's on.

Ah nice workaround, thank you!

carrotIndustries, 2020-06-27:
> One minor problem is that airwire does not get colorized once custom color is assigned- only trace does. Airwire will get assigned color if checked/unchecked or when trace is modified. It seems as if airwires are not getting redrawn after color selection is made.

fixed in 28ff4b20985ed7e43ffcb117ad9e4fef5fa3014d

> The coloring breaks when moving tracks (using m)

fixed in fe89d01f0d2345841df65d6c6cd9d9190133ed87

> Add a way to have the net color be a checkerboard (or similar) pattern,

I've had that in mind for a future enhancement as well. Should the pattern alternate between layer color and set color or set color and transparency?

carrotIndustries, 2020-06-27:
> Making the settings made in the nets window persistent

done in 4e64e05eff9134526d19487ee2c3a63bf6decb63

commit 3accc33c (2020-06-26): imp board: add net colors

closes #408