horizon-eda/horizon #415: Slow part rotation

bob-u, 2020-06-13:
This is a small issue but I think it's worth pointing out:  There is noticeable delay between key press and part rotation.  

Looking at  other EDA software such as EasyEDA, CircuitMaker, LibrePCB and KiCad.  LibrePCB is the fastest- seems instant (maybe too fast), EasyEDA is fast also (but gets slower as you add components) KiCad has a delay but not as big as Horizon EDA.  KiCAD part rotation is probably the slowest I'm comfortable with.

With Horizon EDA I get  a feeling that there is delay between key press and when part starts rotating, for a split second making me question if I pressed the key hard enough.  Horizon EDA can rotate the part quickly when doing arbitrary rotate so it's not rendering that is an issue.

carrotIndustries, 2020-06-13:
I suppose you're referring to the rotate tool rather than pressing <kbd>r</kbd> while moving. In medium-sized designs such as https://github.com/carrotIndustries/x-band-tx/ the rebuild after the tool finishes takes about 120ms. With the latest commit on master, this is reduced to around 40ms depending on which nets are attached to the thing that's moved. How large is the project you're working on?

In order to get Horizon EDA where it is today, some tradeoffs had to made. One of them was to favour correctness and simplicity of implementation over performance. This causes more things to be rebuilt after a tool finishes than strictly necessary since we don't know what a tool actually changed. I already have some ideas on how to improve this.

carrotIndustries, 2020-12-26:
Should be fixed by 06fcd68829cf6acd62b2285efac9d8260444d6a6

closed 2020-12-26 by carrotIndustries