horizon-eda/horizon #416: Strange pin number placement for pads with complex shape

rroohhh, 2020-06-13:
[image]

You can find a pool containing this package here: https://github.com/apertus-open-source-cinema/horizon-pool, its called PQFN-56

bob-u, 2020-06-13:
This is not an error but rather it's just the weird way that Horizon EDA names stacked pads:

[image]

rroohhh, 2020-06-13:
Well it might not be a error, but still something that could be improved.

closed 2020-06-13 by carrotIndustries

bob-u, 2020-06-13:
While we are at it - can the stacked pad text also fade (have the same transparency) with other pads?

[image]

carrotIndustries, 2020-06-13:
> While we are at it - can the stacked pad text also fade (have the same transparency) with other pads?

Nope, the transparency/level of detail is determined by the size of the pad's bounding box. Not sure if using the size of the smallest pad as the level of detail size for all pads in a package is a good idea.

bob-u, 2020-06-14:
Not a big deal, I just like Horizon EDA looks so small things like this bother me.  Is this done for performance reasons?  I would think that transparency should be done based on text size - if it's small to read, it should fade away.

commit ab0272d4 (2020-06-13): fix pad name overlay if bounding box center isn't at origin

fixes #416