horizon-eda/horizon #417: Rotate board view by arbitrary angle

rroohhh, 2020-06-13:
Would be nice, for example to route slanted parts more easily.

carrotIndustries, 2020-06-13:
When revamping the grid I thought about making it possible to specify an angle for the grid but was too lazy to implement it. Would that solve your use case as well?

Anyhow, that won't stop the router from still making 45 degree tracks.

rroohhh, 2020-06-13:
This issue is just about viewing it at an arbitrary angle, the tracks can / should still stick to the normal grid.

pwnorbitals, 2020-12-30:
Seconded, this would be useful

carrotIndustries, 2020-12-31:
See https://github.com/horizon-eda/horizon/tree/canvas-rotate for my take on this. There are still some rough edges such as the grid mark angle, but is this going in the right direction?

rroohhh, 2021-01-02:
I like it! A few notes:
1. I am not a 100% on what to do about the box selection, but I feel it should not be rotate meaning if the view is rotated, the edges should still line up with the window borders.
2. It would be nice to be able to select the origin of the rotation. One example where I would like to use this feature is to route a IC rotated by 45 degrees to one not and it would be nice to easily switch between a view aligned with the IC rotated by 45 degrees and a view aligned with the other IC. If the ICs are located far from the origin, they will move a lot and I need to pan around to get them on my screen again.

carrotIndustries, 2021-01-03:
> the edges should still line up with the window borders.

Of course they should. I was so occupied with getting the in-pad text and grid to look right that I didn't try selecting anything.

Both selection and grid should now work correctly.

> It would be nice to be able to select the origin of the rotation.

I'm not entirely happy with the current situation either. Rotating around the viewport will probably cover your use case.

How should arrow keys behave with regards to the rotated canvas? Should pressing down move the selection in negative y direction on the screen regardless of the rotation?

carrotIndustries, 2021-01-03:
The latest commit moved the angle controls to the view options menu. With this, only 45 degree increments are supported as I couldn't think of a way to fit the spinbutton in a nice way. Any ideas where to put such a spinbutton, perhaps in a separate window?
[image]

rroohhh, 2021-01-03:
What about just making the text with the angle editable?

carrotIndustries, 2021-01-03:
> What about just making the text with the angle editable?

Clicking the button with the angle in it resets the angle to 0°. The scale control in nautilus works the same way. Adding a reset button would disrupt the symmetry of the arrangement.

carrotIndustries, 2021-01-03:
Some math later, flipping the view works when the canvas is at an angle.

rroohhh, 2021-01-04:
> > What about just making the text with the angle editable?
> 
> Clicking the button with the angle in it resets the angle to 0°. The scale control in nautilus works the same way. Adding a reset button would disrupt the symmetry of the arrangement.

I see, I guess I would be fine doing something more complex aswell, like keeping the angle a button, that resets the angle when you click it with the left mouse button and turns in a textfield if you click it with the right mouse button for example.

carrotIndustries, 2021-01-05:
From my perspective, the the implementation on the https://github.com/horizon-eda/horizon/tree/canvas-rotate branch is ready for merging.

I'm still a bit on the edge if we should merge this before the 1.4 release due end of this month, or wait a while so that this feature gets some more testing before making it into a release.

commit d69b69c5 (2021-01-08): imp: add canvas rotation

closes #417
closed 2021-01-08 by carrotIndustries