horizon-eda/horizon #419: Rule to prevent via in pad

rroohhh, 2020-06-13:
Currently there is no way (atleast I did not find one) to specify the minimum clearance between a via and a pad of the same net.

carrotIndustries, 2020-06-14:
Should this clearance be based on via hole to pad distance or via pad to pad distance?

rroohhh, 2020-06-14:
Via pad to pad distance seems better to me.

carrotIndustries, 2020-06-14:
Okay, seems reasonable. The more general solution here is same net clearance. However, there's an issue with that:

For DRC, the entire board gets dumped into a set of polygons (patches) grouped by type (via, track, etc), layer and net. This means that we can't do same-net clearance between to objects of the same type as they could have been merged into one patch if they overlap.

Can you think of any other sensible type pairs other than pad-via for same net clearance?

rroohhh, 2020-06-15:
Some I think could be useful:
- via to via clearance
- via to pth clearance
- pad to pth clearance

carrotIndustries, 2020-06-28:
Have a look at the commit mentioned above. If it works as intended, I'll merge it into master.

rroohhh, 2020-06-29:
If I understand the previous commit correctly, this only applies those rules during DRC, and not for example for the router, right?

I think having the router abide by these rules would be very useful.

carrotIndustries, 2020-06-29:
> If I understand the previous commit correctly, this only applies those rules during DRC, and not for example for the router, right?

Yes, as far as I am aware the kicad router doesn't support same net clearance.

rroohhh, 2020-06-29:
I see. 

I will play around with it a bit more.

commit 46c2156b (2020-08-29): add same net clearance check

fixes #419
closed 2020-08-29 by carrotIndustries