horizon-eda/horizon #422: Ability to specify router attachment points for complex padstacks

rroohhh, 2020-06-15:
For a padstack like this:
[image]
it would be useful to have some way of specifying where the router should attach to.
 
Currently this seems to work by using the origin, but this means only one attachment point is possible. It would be nice to be able to specify three, one for each spoke.

bob-u, 2020-06-15:
I have not used stacked padstack, but what you describe would be my expectation as well.

rroohhh, 2020-06-15:
Didn't try drawing a polygon, but using `Draw track` I can of course get the track where I want it.

carrotIndustries, 2020-06-20:
Not sure if this is really worth the added complexity. Setting the padstacks origin somewhere sensible and connecting the other sub-pads using draw tracks to be an acceptable workaround to me.

rroohhh, 2020-06-22:
I don't think this a very important feature, however I think parts with padstacks like this are common enough that some better solution that manually using draw tracks would be very nice and contribute to the overall feeling of polish a lot.

bob-u, 2020-06-22:
Out of curiosity, can you point to a datasheet that requires pad shape like in the screenshot where copper joining pads is smaller than pad itself?  Normally copper area is of full length of single pin pad and the reason they are joined is for thermal or resistive heating.  Therefore, perhaps the footprint needs fixing?

carrotIndustries, 2020-06-22:
Not this part, but especially power ICs tend to come in weird packages like this one: https://www.infineon.com/dgdl/ir3846m.pdf?fileId=5546d462533600a4015355d568a417ee

[image]

rroohhh, 2020-06-22:
> Out of curiosity, can you point to a datasheet that requires pad shape like in the screenshot where copper joining pads is smaller than pad itself? Normally copper area is of full length of single [...]

I don't think the footprint needs fixing, I just drew what the [datasheet](https://www.infineon.com/dgdl/Infineon-IRPS5401M-DataSheet-v02_04-EN.pdf?fileId=5546d4625cc9456a015cd69d402139db) recommended:
[image]

bob-u, 2020-06-23:
I think the footprint shape is designed this way for silkscreen and paste application.  Would you agree that one should apply copper pour to the whole width of the pad stack?  Therefore, trace attachment point would not be that important since it would be covered by a polygon?

rroohhh, 2020-06-23:
> I think the footprint shape is designed this way for silkscreen and paste application. Would you agree that one should apply copper pour to the whole width of the pad stack? Therefore, trace attachment point would not be that important since it would be covered by a polygon?

I am no expert, but the datasheet has a seperate footprint for the paste application and specifically labels the imaged footprint as `PCB PAD`. Also the altium footprint one can download from their website has the same copper layer as my footprint. So atleast infineon seems to think this is how it should be done.

If it actually is necessary I don't know, probably not.

unrelentingtech, 2022-01-11:
Having multiple routing attachment points would also be useful for an [SMD-and-TH combo padstack](https://github.com/horizon-eda/horizon/pull/644#issuecomment-1005092881).

carrotIndustries, 2022-03-23:
See the linked branch for an implementation. Rather than having the package/padstack specify where the pad attaches, the user can freely select the connection point using the "Move track connection" tool. Let me know if that works for you.

commit 21840a72 (2022-04-01): add track connection offset

closes #422
closed 2022-04-01 by carrotIndustries