horizon-eda/horizon #425: Feature request: place group of unplaced components on the board with one click

bob-u, 2020-06-15:
Components come in different sizes and I would like to be able to select a group of unplaced components and place them on the board in one go so I can then move them around.  Placing the serially, one by one, feels a bit like playing Tetris where you don't know what the next component will be and which orientation and location I should place the current one.

As a bonus feature, it would be nice if Unplaced components had package specifications along with them, either as a name or more fancy, footprint rendering.

carrotIndustries, 2020-06-15:
So you'd expect them to be arranged semi-randomly within that group in way that they don't overlap?

bob-u, 2020-06-15:
I would not mind just a pile of packages to drop at my chosen location (saves on mouse clicks and gives me something to work with visually.  

If they were arranged randomly would be better and if components where to mimic orientation in the schematic might be the best (though it depends on type of the schematic).