horizon-eda/horizon #426: Feature request: keyboard shortcuts for setting visibility of all layers

jue89, 2020-06-15:
I like to go through all layers one-by-one for a final check of boards and packages.
To achieve this, I disable the visibility of all layers by clicking all the eye icons and then step through them with the arrow keys. The highlighted layers becomes visible temporary.

Changing the visibility of all layers is a tedious mouse-driven task.

It would help to have keyboard shortcuts to:
* disable the visibility of all layers
* enable the visibility of all layers

carrotIndustries, 2020-06-15:
I think I have a better way to address this issue: Add some kind of dropdown menu to the layer box on the left with these options
 - Display layers as-is
 - Work (selected) layer only
 - Shadow other layers (similar to kicad's high contrast mode)

If desired, we can add actions for these options so that they're reachable using shortcuts.

jue89, 2020-06-16:
> I think I have a better way to address this issue: Add some kind of dropdown menu to the layer box on the left with these options
>
>  - Display layers as-is
>
>  - Work (selected) layer only
>
>  - Shadow [...]

Sounds great :) Would love to see this feature in the future.

bob-u, 2020-06-16:
On same topic, I would like to see pads more prominent - they get lost under copper color.

[image]

[image]

[image]

closed 2020-06-20 by carrotIndustries

bob-u, 2020-06-20:
[image]

Looks good.  Are you planning to add keyboard shortcuts at later timet?

jue89, 2020-06-21:
Love it. Great work, @carrotIndustries!

commit 703ea35e (2020-06-20): add work layer only display modes

closes #426