horizon-eda/horizon #428: FYI: Jun15 build is much slower than v1.1.1 realease

bob-u, 2020-06-15:
Just fyi, when dragging a package I get 'render took 97.3455' where as v1.1.1 shows 'render took 74.096'.  Anti-aliasing setting has no effect on the render time, except that it takes ~~50 (Jun15) vs ~35(v1.1.1) to render when switching it on/off.

There is noticeable difference, between versions mentioned.  With Jun15 build zoom is jerky (I can see block of the screen moving out of sync the rest), panning makes big jumps, crosshair is slow to catch up with the pointer.

carrotIndustries, 2020-06-15:
> 97.3455' where as v1.1.1 shows 'render took 74.096

That output is a bit misleading as it's the inverse of the time it took to render to get a better idea of the framerate: https://github.com/horizon-eda/horizon/blob/master/src/canvas/render.cpp#L1492 So higher numbers are better.

It's expected that antialiasing doesn't effect that output since that number only measures the time it took to translate the board into something that the GPU can render. Antialiasing will mostly affect zoom/pan.

>  With Jun15 build zoom is jerky (I can see block of the screen moving out of sync the rest)

Can you share a screenshot of what this looks like? As I can't reproduce this, I need to know the commit that introduced this bug.

Are you on windows or on linux? The windows zips provided by the CI always include the latest dependencies, so one of them might have triggered this behaviour as well.

bob-u, 2020-06-15:
It could be some issue with my system as I switched from Intel 620 to AMD Vega 8 today.  I expected to see a speed increase as KiCad is much smoother on this desktop (GPU is twice as fast) but got the opposite.

I built from current source (git clone) and from v1.1.1 source freshly installed Ubuntu 20.04 LTS.  I run OpenGL 4.5 benchmark, Basemark GPU and got 709 score which is in line with other results for same GPU.

carrotIndustries, 2020-06-15:
You might also try other applications that use Gtk and OpenGL such as solvespace. Even with an intel GPU, I've had some combinations of mesa and xf86-video drivers that work so well and caused glitchy rendering.

bob-u, 2020-06-15:
[image]

I think you're right - this a system issue.  Above is what happens when resizing window - refresh rate is very low and it does not update window completely sometimes.

bob-u, 2020-06-16:
Downgrading to 18.04.04 fixed the issue.  For the record, I am also using kisak-mesa PPA.

closed 2020-06-16 by bob-u