horizon-eda/horizon #429: Note: Jerky cursor move /  redraw rate depends on platform

bob-u, 2020-06-16:
Difference in platforms: One is smoother than the other, both Ubuntu 18.04.04, Mesa 20.1.1.

1.  Intel HD 620
[](https://streamable.com/phyhty)

2. AMD Vega 8

carrotIndustries, 2020-06-16:
Does this also happen if the canvas is empty or in the symbol editor where there's almost nothing to render? If yes, the problem is likely to be elsewhere (Gtk, mesa, X11) in the stack.

bob-u, 2020-06-16:
I tested with your demo X-band board.  If there's any difference between all clear and with the board present - it's hard to notice.  Jerkiness  is best to reproduce moving the mouse diagonally, probably because x and y jerkiness  add up.  (I am not claiming that it's the program, just reporting).

bob-u, 2020-07-14:
On same machine on Windows (same result with Intel, nVidia and AMD GPUs) when dragging rendering of lines  dos not change.  On Linux, rendering of lines (or aliasing) seems to change every step.  I am curious if this is a function of GTK or something else?

Line changes thickness and brightness as it's moved up:

[image]

bob-u, 2020-07-14:
Windows render:

[image]

carrotIndustries, 2020-07-14:
> Line changes thickness and brightness as windows contents are dragged up 

Works for me.

> GTK or something else?

Probably GPU drivers/mesa. I don't see how this could happen with no antialiasing.

bob-u, 2020-07-14:
> Works for me.

Are you on Windows 10?

> Probably GPU drivers/mesa. I don't see how this could happen with no antialiasing.

It looks the same on Intel, AMD and nVidia GPUs therefore I doubt it's a driver.  I double-checked that anti-aliasing is off.  

Even with MSAA off anti-aliasing appears to be still on for lines and probably other primitives so I would guess this is something to do with how vectors are rendered.

carrotIndustries, 2020-07-17:
> Are you on Windows 10?

Nope, X11 with intel GPU.