horizon-eda/horizon #430: How to move up and down in 3D preview?

bob-u, 2020-06-17:
On multi-layer boards, once layers are 'exploded' how do I move the view-point up and down to zoom at different levels of traces?

carrotIndustries, 2020-06-20:
The commit above fixes this by translating up/down cursor movement to shifting the z component of the point the camera looks at depending on the cam's elevation.

Unfortunately this makes rotating a bit awkward after rotating back to top view and it breaks my concept of having control over all view parameters (center xyz, azimuth, elevation) in all cases.

I remember there being name for the way of mapping mouse input to 3D view changes, but it won't come to my mind...

bob-u, 2020-06-20:
Hm, I am probably misunderstanding your description - for me up/down arrow explodes the layers.

I found that mouse movement with SHIFT+LMB pans up/down left/right.   Without the shift it rotates.

It works great and it's quite intuitive and precise.  Great work!

carrotIndustries, 2020-06-23:
> I found that mouse movement with SHIFT+LMB pans up/down left/right. Without the shift it rotates.
> It works great and it's quite intuitive and precise. Great work!

The primary way for panning is middle mouse button. The shift key only got added to pan without a middle mouse button.

So can we close the issue?

bob-u, 2020-06-23:
Yes, it works great.

closed 2020-06-23 by bob-u