horizon-eda/horizon #431: Horizon flatpack cannot open schematic or board due to Glib error

bob-u, 2020-06-18:
I get this Glib  when trying to start Horizon installed through Flatpak and program exits with code 134:

[image]

bob-u, 2020-06-18:
xdpyinfo:
ame of display:    :0
version number:    11.0
vendor string:    The X.Org Foundation
vendor release number:    12008000
X.Org version: 1.20.8
maximum request size:  16777212 bytes
motion buffer size:  256
bitmap unit, bit order, padding:    32, LSBFirst, 32
image byte order:    LSBFirst
number of supported pixmap formats:    7
supported pixmap formats:
    depth 1, bits_per_pixel 1, scanline_pad 32
    depth 4, bits_per_pixel 8, scanline_pad 32
    depth 8, bits_per_pixel 8, scanline_pad 32
    depth 15, bits_per_pixel 16, scanline_pad 32
    depth 16, bits_per_pixel 16, scanline_pad 32
    depth 24, bits_per_pixel 32, scanline_pad 32
    depth 32, bits_per_pixel 32, scanline_pad 32
keycode range:    minimum 8, maximum 255
focus:  window 0x4800007, revert to Parent
number of extensions:    28
    BIG-REQUESTS
    Composite
    DAMAGE
    DOUBLE-BUFFER
    DPMS
    DRI2
    DRI3
    GLX
    Generic Event Extension
    MIT-SCREEN-SAVER
    MIT-SHM
    Present
    RANDR
    RECORD
    RENDER
    SECURITY
    SHAPE
    SYNC
    X-Resource
    XC-MISC
    XFIXES
    XFree86-DGA
    XFree86-VidModeExtension
    XINERAMA
    XInputExtension
    XKEYBOARD
    XTEST
    XVideo
default screen number:    0
number of screens:    1

screen #0:
  dimensions:    2160x3840 pixels (572x1016 millimeters)
  resolution:    96x96 dots per inch
  depths (7):    24, 1, 4, 8, 15, 16, 32
  root window id:    0x6d7
  depth of root window:    24 planes
  number of colormaps:    minimum 1, maximum 1
  default colormap:    0x20
  default number of colormap cells:    256
  preallocated pixels:    black 0, white 16777215
  options:    backing-store WHEN MAPPED, save-unders NO
  largest cursor:    128x128
  current input event mask:    0xda0003
    KeyPressMask             KeyReleaseMask           StructureNotifyMask      
    SubstructureNotifyMask   SubstructureRedirectMask PropertyChangeMask       
    ColormapChangeMask

carrotIndustries, 2020-06-18:
I remember having seem this on my system (intel GPU) a while back a well. Try some other flatpak app that uses GtkGLArea such as hexgl and see if that app crashes as well (it did in my case).

To me, this seems more of a flatpak issue than anything else.

bob-u, 2020-06-18:
HexGl freezes whole desktop as you predicted.  Horizon-EDA built from source works just fine (except for the slow cursor updates I mention in another ticket).

closed 2020-06-18 by bob-u