horizon-eda/horizon #434: Differential pair router only likes horizontal and vertical vias

rroohhh, 2020-06-22:
The p&s router puts the vias for differential pairs either on the same x or same y coordinate. This leads to strange via placement for parts at an angle.

[image]

I think it should atleast allow for 45 degree angles aswell.

carrotIndustries, 2020-06-22:
> I think it should atleast allow for 45 degree angles aswell.

It does in some places. As the router is more or less a black box to me, I don't think there's much I can do about it...

rroohhh, 2020-06-22:
Right, I wasn't sure, if this was a thing the router does or something horizon does.

I guess I can try digging around a bit in the router.

bob-u, 2020-06-22:
Maybe changing grid orientation, as @carrotIndustries mentioned might be possible, could change router's via orientation also.

phkahler, 2021-01-10:
I wonder if that's trying to maintain equal lengths as well. obviously putting them on a 45 would allow perfect symmetry on this side of the board, but maybe it's part of a larger solution involving another layer.