horizon-eda/horizon #436: length tuning is very slow

rroohhh, 2020-06-23:
Length tuning can be very slow, especially when trying to tune a track bordered by two other tracks like the track marked red:
[image]

(A board with tracks aranged like this can be found [here](https://github.com/apertus-open-source-cinema/axiom-micro-mainboard/commit/6f3c0c562a0489529584d1911f43c8a990ad20e8), simply replace one of the length matched tracks with a straight one and try to length match it again)

carrotIndustries, 2020-06-23:
I was able to reproduce it and came to the conclusion that it's probably the kicad router being slow as the flamegraph mostly is just router functions. Replacing my clearance lookup with returning a constant made no difference.

It'd be interesting to know if the router behaves similarly in kicad.

rroohhh, 2020-06-23:
I obviously don't have a exact match, but playing around a bit with it in kicad it feels way faster.