horizon-eda/horizon #437: Schematic grid setting is 'grayed out'

bob-u, 2020-06-23:
Schematic's grid dimensions cannot be changed:

[image]

carrotIndustries, 2020-06-23:
This is intended behaviour since the schematic's grid has to match the pin grid of the symbols in the pool. Changing the grid will lead to people messing up their schematics and cause even more issues.

closed 2020-06-23 by carrotIndustries

katyo, 2020-10-15:
@carrotIndustries But the ability to change grid still matter for symbol editing.

carrotIndustries, 2020-10-15:
> But the ability to change grid still matter for symbol editing.

Why? with a user-definable grid users might wind up placing pins off-grid. If you need a finer grid for drawing the symbol itself, you can always switch to the fine grid by pressing alt/ctrl.

rapgenic, 2021-03-25:
What about using a fully custom library with a custom pin to pin spacing? 

For example, in my use case I want to design my own library, and for stilistic (historic) reasons I'd use a 100mil grid for both the library and the schematic.