horizon-eda/horizon #438: Feedback: Copper pour/ polygon editing

bob-u, 2020-06-24:
I'm having a hard time laying copper pours in Horizon EDA mainly because polygon edges are not visible.  I know there's a way to convert polygon to outline so edges are visible, however if I do that I loose Net information and all other plane properties.

In summary, if anyone asked me what is the trickiest and time consuming task in Horizon EDA I would say that it's laying copper fills.

carrotIndustries, 2020-06-28:
Since latest commits, all edges of a polygon are show if one i selected:
[image]

Does this address your issue?

bob-u, 2020-06-30:
Nice improvement.  One issue I had was that copper plane may have had very little visible features that could be clicked on to select it's polygon.  Hover select solves it - just press escape to activate and move mouse cursor around the plane for polygon edges/vertices to be revealed.  

What is not easy is making precise positional adjustments to vertices or edges because cursor is snapping to objects.

Below is a list of enhancements that I've been thinking about.  Of those, I would consider 2, 5 a  higher priority. 

1. Edge marker to have different shape than vertex (so it's easier to pick when markers are close to each other)

2. Suppress cursor snapping when moving vertex or edge ( would be useful for other move tasks, such as moving a component) For example, if ALT selects fine grid, SHIFT to suppress snapping, ALT+SHIFT fine movement and no snapping.

3. When vertex marker is clicked on directly (within marker), to save on clicks, do not ask if user wants edges the vertex belongs to, just select the vertex.

4. Ability to lock axis when moving - when pressed a certain key (ctrl, alt, etc)  cursor is locked to x or y axis (depending on the key or locks to initial direction of the movement). 

5. Ability to set transparency for 'Selectable' markers or borders (right now markers cover up what you're selecting, especially at lower zoom levels.

6. Ability to adjust area within which select-able object list is created and presented to user for final selection  (as a bonus, make it zoom level sensitive - area gets smaller as zoom is higher).

7. Make it select-able which objects the cursor may snap to.

8. Display priority when polygon is selected in the info panel.

carrotIndustries, 2020-06-30:
> Ability to lock axis when moving - when pressed a certain key (ctrl, alt, etc) cursor is locked to x or y axis (depending on the key or locks to initial direction of the movement).

Press slash to restrict axis in the move tool.

> When vertex marker is clicked on directly (within marker), to save on clicks, do not ask if user wants edges the vertex belongs to, just select the vertex.

Use hover select for this case.

> Ability to adjust area within which select-able object list is created and presented to user for final selection 

Don't get it can you, elaborate on this?

bob-u, 2020-07-01:
> Press slash to restrict axis in the move tool.

Good to know!  It does take little to get used to but it works.  Based on experience with other programs, I expected for key to lock axis wherever the object currently is but the way it works in horizon it locks to original object location and also only one axis at a time.  In Horizon one has to move one axis, save it, move another axis.

> Use hover select for this case.

That indeed works!  I would had preferred for this to happen in all circumstances but it's a low priority.

> 

    Ability to adjust area within which select-able object list is created and presented to user for final selection

Don't get it can you, elaborate on this?

Click to select offers a menu of near-by objects if there's more than one within certain distance from the click.  Make this certain distance configurable.