horizon-eda/horizon #439: No microSD card in cardreader causes error popup on launch

VasiliSker, 2020-06-24:
I have win 7 empty USB card reader connected, when launched Horizon it pops 'no drive' error, and keep showing it no matter what i click (ignore/continue/repeat)

bob-u, 2020-06-24:
Take a look at: https://www.easeus.com/partition-manager-software/fix-no-disk-in-drive-error.html#3

Try fix 3 or 6.

VasiliSker, 2020-06-24:
thanks, 6 should work but i would suggest to not spam all hard drives with requests that will cause that

bob-u, 2020-06-24:
Certainly.  I just posted a workaround so it does not stop you or someone else from exploring the software until it's fixed.

carrotIndustries, 2020-06-24:
Hm, interesting. I guess this a Gtk issue. Unfortunately, I can't reproduce this issue. What does sysinternals procmon show for horizon-eda.exe?

bob-u, 2020-06-25:
Something similar is discussed here: https://github.com/Beep6581/RawTherapee/issues/3544

carrotIndustries, 2020-06-25:
Try https://ci.appveyor.com/project/carrotIndustries/horizon/builds/33727952 once the build has finished and see if it fixes your issue.

VasiliSker, 2020-06-25:
sorry cant reproduce after setup. Happened only on first launch

bob-u, 2020-06-25:
> I have win 7 empty USB card reader connected, when launched Horizon it pops 'no drive' error, and keep showing it no matter what i click (ignore/continue/repeat)

> sorry cant reproduce after setup. Happened only on first launch

Was the fix to just restart the app and issue went away w/o making changes?

VasiliSker, 2020-06-25:
Applied fix from comment 2, launched horizon and downloaded tools, after that it works even if fix deleted. also at first start it showed some different intro screen

VasiliSker, 2020-06-25:
Fix works, no errors pops out.

commit 9b0d1e94 (2020-06-25): win32: fix annoying drive not ready errors

should fix #439
closed 2020-06-25 by carrotIndustries