horizon-eda/horizon #440: Routing offset settings seem to get lost

rroohhh, 2020-06-26:
After routing some tracks (using the router) it seems like the design rules change (obviously without actually changing them). Opening the rules window and pressing "Apply rules" doesn't change anything.

It seems like the routing offset setting gets lots. (As I can get back to the unchanged design rules my manually specifying the clearance offset when routing a track)

I am not really sure on how to track this down further, let me know if you have some ideas.

bob-u, 2020-06-26:
Since router was recently updated - are you running latest build?  Can you check out one of the builds before router update to see if the issue is there?

rroohhh, 2020-06-26:
> Since router was recently updated - are you running latest build? Can you check out one of the builds before router update to see if the issue is there?

Yes I am running the latest commit, trying some older versions now.

rroohhh, 2020-06-26:
Ok I tried commit ab0272d4 (just before the router update) and I can't reproduce with that, so it it seems to be connected to that.

carrotIndustries, 2020-06-26:
I tried reproducing it, but didn't have any success. For me I couldn't tell a difference between specifying the clearance offset manually or using the ones from the rules.

rroohhh, 2020-06-26:
Hmm, I played around a bit more and undo seems to be a trigger, after drawing a track and then undoing a couple of times the clearance offset from the rules seems to get lost.

(For reference: I tried it on [this](https://github.com/apertus-open-source-cinema/axiom-micro-mainboard/commit/f17ab0801a27797fefb95f6ed7bddd2409abc190) commit, routing a track from the `RAM_DQ0` via through the 3x3 array of vias.)

closed 2020-06-27 by carrotIndustries

rroohhh, 2020-06-27:
Thank you, that seems to have fixed it (atleast I can't reproduce it anymore).

commit d251b965 (2020-06-27): fix memory corruption in rules

rules get copied from board to core but the pointers to the sorted rules
still pointed to the deleted temporary

should fix #440