horizon-eda/horizon #442: Key configuration for tool actions

rroohhh, 2020-06-26:
Currently it seems like there is no way to configure the keys a tool uses (like changing the key "route track" uses to place a via and so on).

carrotIndustries, 2020-07-07:
There are two ways i think this could work out:

 - Each tool defines actions, add UI to map keys to these actions. If a user wasn't happy with r for rotate, that'd need to be changed in all tools then.
 - Define a set of in-tool actions globally, set keys per action, so if setting a different key for rotate, all tools will be consistent.

rroohhh, 2020-07-08:
Hmm, while in most cases I would assume having the in-tool actions global and thus always having consistent keybindings for all tools, I wonder if there are cases where it would maybe be nicer to for example map keys close to each other to be able to press them more easily. However given that I currently can't think of any specific example for that, I think having global in-tool actions and consistent key bindings seems fine for now.

carrotIndustries, 2020-07-09:
> I think having global in-tool actions and consistent key bindings seems fine for now.

I tried to get an inventory of keybindings we currently have in tools and discovered that while there are quite a lot of things that would make sense to have globally such as mirror, rotate, set width there's also a sizable number of keybindings that are very specific to tools such as cycling dimension modes or changing the size of a net label while drawing a net.

Perhaps it'd make sense to group these into common and tool-specific keybindings in the UI.

I also don't have an idea on how to keep a concise representation in the tool tip bar for actions such as net label size that are currently bound to +/-. Just putting the whole thing as in `+,=: increase net label size -:decrease net label size`in the bar isn't really what I want to do.

rroohhh, 2020-07-10:
> Perhaps it'd make sense to group these into common and tool-specific keybindings in the UI.

Yeah makes sense.

> I also don't have an idea on how to keep a concise representation in the tool tip bar for actions such as net label size that are currently bound to +/-. Just putting the whole thing as in `+,=: increase net label size -:decrease net label size`in the bar isn't really what I want to do.

Maybe combining common options like `x,=/-: increase / decrease net label size` could work?

carrotIndustries, 2020-07-12:
Currently, conceptually similar in-tool actions such as flip arc and mirror both use <kbd>e</kbd>. Should we have one in-tool action mirror that's used for both mirroring and flipping arcs, or should there be a mirror and a flip arc action?

rroohhh, 2020-07-13:
I would say seperate actions. 
One (usually) needs to change the keybinding for a action only once, so having to change two is not a lot of additional work, so I think the added flexibility of making them seperate is worth it.

carrotIndustries, 2020-07-14:
While we're at it: There are some in-tool actions that have an equivalent action, such as rotate or mirror. Should the in-tool rotate action be configurable independently from the regular rotate action?

@atoav I took this as an opportunity to implement your suggestion regarding the appearance of the tool bar:
[image]

rroohhh, 2020-07-15:
> While we're at it: There are some in-tool actions that have an equivalent action, such as rotate or mirror. Should the in-tool rotate action be configurable independently from the regular rotate action?

I would say making the in-tool action and the equivalent "normal" action not independent is fine.
 
> @atoav I took this as an opportunity to implement your suggestion regarding the appearance of the tool bar:
> [image]

Wow, that looks very nice!

carrotIndustries, 2020-07-16:
See branch https://github.com/horizon-eda/horizon/tree/in-tool-action-rebased. I'm not sure if it's a good idea to merge it into master yet as we'll have the next release in two weeks and I'd like to see some more user testing since I had to touch all tools in order to adapt them to the new in-tool actions.

bob-u, 2020-07-16:
I had to remove second argument to Gtk::Image in imp.cpp in order to compile.  I think because of it images of mouse were not visible.  It looks very good.  I only had few minutes to look, one nit-pick is that it was hard to tell upper case from lower case  of letters which height is the only differentiation factor, lie S and s for example.  Some kind of indication of SHIFT would be helpful.

carrotIndustries, 2020-07-16:
> I had to remove second argument to Gtk::Image in imp.cpp in order to compile

Which distro are you using?

bob-u, 2020-07-16:
> Which distro are you using?

Ubuntu 18.04.4 LTS
|/ Name                        Version            Architecture       Description
+++-===========================-==================-==================-============================================================
ii  libgtk-3-0:amd64            3.22.30-1ubuntu4   amd64              GTK+ graphical user interface library
ii  libgtk2.0-0:amd64           2.24.32-1ubuntu1   amd64              GTK+ graphical user interface library

bob-u, 2020-07-16:
pc2@pc2-desktop:~/horizon$ make -j 5
XMLLINT not set and xmllint not found in path; skipping xml preprocessing.
 build/gen/resources.cpp
 build/obj/build/gen/version_gen.o
XMLLINT not set and xmllint not found in path; skipping xml preprocessing.
 build/gen/help_texts.cpp
 build/obj/src/imp/imp.o
 build/obj/src/imp/imp_package_3d.o
 build/obj/src/imp/imp_board.o
 build/obj/src/imp/imp_frame.o
 build/obj/src/imp/tool_popover.o
 build/obj/src/imp/selection_filter_dialog.o
src/imp/imp.cpp: In member function ‘void horizon::ImpBase::tool_bar_append_action(horizon::InToolActionID, horizon::InToolActionID, const string&)’:
src/imp/imp.cpp:2254:68: error: no matching function for call to ‘Gtk::Image::Image(Gtk::BuiltinIconSize)’
         auto img = Gtk::manage(new Gtk::Image(Gtk::ICON_SIZE_BUTTON));
                                                                    ^
In file included from /usr/include/gtkmm-3.0/gtkmm/entry.h:33:0,
                 from /usr/include/gtkmm-3.0/gtkmm/treeview.h:41,
                 from /usr/include/gtkmm-3.0/gtkmm/combobox.h:33,
                 from /usr/include/gtkmm-3.0/gtkmm/appchooserbutton.h:27,
                 from /usr/include/gtkmm-3.0/gtkmm.h:107,
                 from src/core/clipboard.hpp:6,
                 from src/imp/imp.hpp:2,
                 from src/imp/imp.cpp:1:
/usr/include/gtkmm-3.0/gtkmm/image.h:251:12: note: candidate: Gtk::Image::Image(const Glib::RefPtr<Gdk::PixbufAnimation>&)
   explicit Image(const Glib::RefPtr<Gdk::PixbufAnimation>& animation);
            ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:251:12: note:   no known conversion for argument 1 from ‘Gtk::BuiltinIconSize’ to ‘const Glib::RefPtr<Gdk::PixbufAnimation>&’
/usr/include/gtkmm-3.0/gtkmm/image.h:248:3: note: candidate: Gtk::Image::Image(const Glib::RefPtr<Gtk::IconSet>&, Gtk::IconSize)
   Image(const Glib::RefPtr<IconSet>& icon_set, IconSize size);
   ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:248:3: note:   candidate expects 2 arguments, 1 provided
/usr/include/gtkmm-3.0/gtkmm/image.h:242:3: note: candidate: Gtk::Image::Image(const Gtk::StockID&, Gtk::IconSize)
   Image(const Gtk::StockID& stock_id, IconSize size);
   ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:242:3: note:   candidate expects 2 arguments, 1 provided
/usr/include/gtkmm-3.0/gtkmm/image.h:225:14: note: candidate: Gtk::Image::Image(const Glib::RefPtr<Gdk::Pixbuf>&)
     explicit Image(const Glib::RefPtr<Gdk::Pixbuf>& pixbuf);
              ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:225:14: note:   no known conversion for argument 1 from ‘Gtk::BuiltinIconSize’ to ‘const Glib::RefPtr<Gdk::Pixbuf>&’
/usr/include/gtkmm-3.0/gtkmm/image.h:218:14: note: candidate: Gtk::Image::Image(const string&)
     explicit Image(const std::string& file);
              ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:218:14: note:   no known conversion for argument 1 from ‘Gtk::BuiltinIconSize’ to ‘const string& {aka const std::__cxx11::basic_string<char>&}’
In file included from /usr/include/gtkmm-3.0/gtkmm/entry.h:33:0,
                 from /usr/include/gtkmm-3.0/gtkmm/treeview.h:41,
                 from /usr/include/gtkmm-3.0/gtkmm/combobox.h:33,
                 from /usr/include/gtkmm-3.0/gtkmm/appchooserbutton.h:27,
                 from /usr/include/gtkmm-3.0/gtkmm.h:107,
                 from src/core/clipboard.hpp:6,
                 from src/imp/imp.hpp:2,
                 from src/imp/imp.cpp:1:
/usr/include/gtkmm-3.0/gtkmm/image.h:206:3: note: candidate: Gtk::Image::Image()
   Image();
   ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:206:3: note:   candidate expects 0 arguments, 1 provided
/usr/include/gtkmm-3.0/gtkmm/image.h:169:12: note: candidate: Gtk::Image::Image(GtkImage*)
   explicit Image(GtkImage* castitem);
            ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:169:12: note:   no known conversion for argument 1 from ‘Gtk::BuiltinIconSize’ to ‘GtkImage* {aka _GtkImage*}’
/usr/include/gtkmm-3.0/gtkmm/image.h:168:12: note: candidate: Gtk::Image::Image(const Glib::ConstructParams&)
   explicit Image(const Glib::ConstructParams& construct_params);
            ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:168:12: note:   no known conversion for argument 1 from ‘Gtk::BuiltinIconSize’ to ‘const Glib::ConstructParams&’
/usr/include/gtkmm-3.0/gtkmm/image.h:152:3: note: candidate: Gtk::Image::Image(Gtk::Image&&)
   Image(Image&& src) noexcept;
   ^~~~~
/usr/include/gtkmm-3.0/gtkmm/image.h:152:3: note:   no known conversion for argument 1 from ‘Gtk::BuiltinIconSize’ to ‘Gtk::Image&&’
Makefile:846: recipe for target 'build/obj/src/imp/imp.o' failed
make: *** [build/obj/src/imp/imp.o] Error 1
make: *** Waiting for unfinished jobs....

carrotIndustries, 2020-07-16:
> Some kind of indication of SHIFT would be helpful, maybe underline.

How about this? @rroohhh @atoav any opinions?

[image]

bob-u, 2020-07-16:
This is a small issue but since we are on the topic: IMO, aesthetically single letter w/o shift icon looks better and takes up less space.  Therefore, maybe using larger upper case letter would be enough of a hint, of make a small bump in the side walls as a  reference to better judge font size:

[image]

There is advantage to have separate icon for shift.   [shift]+i and l is easy to tell apart I and l are not.  In addition shift icon + letter could be expanded into alt + letter, etc.  which is more universal and probably better for high resolution displays.   If shift icon is used, letter should be lower case, uppsercaset seems redundant.

carrotIndustries, 2020-07-19:
Closing since it's on master now.

closed 2020-07-19 by carrotIndustries