horizon-eda/horizon #443: Add instructions for length tuning when dialog is empty

Timmmm, 2020-06-27:
Hey, having a bit more of a play with Horizon and last time I did you were amazing at fixing the UX paper cuts. Here is another one - I'm trying to route differential pairs so I opened the "Length Tuning" dialog, but it is empty. I can't figure out how to get it to show any tracks. It would be cool if when empty it showed a message that told you how to do it. Something like this:

[image]

(I don't know what the actual instructions are yet!)

carrotIndustries, 2020-06-27:
Which version are you running? Since 38044b2b99d795f05f3f359866f9a388b6de3851 the window should look like this when empty:
[image]

Timmmm, 2020-06-27:
Aha! I am running an older version; I'll update. Nice one for already doing this!!

closed 2020-06-27 by Timmmm

Timmmm, 2020-06-27:
Although maybe one small clarification would help (assuming this behaviour isn't just in 1.1.0):

"Select some tracks and then use "Measure (all) tracks" to add tracks to this list."

I can't figure out what the difference between "Measure track" and "Measure all tracks" is - both seem to only measure the selected tracks and both are unavailable when no tracks are selected. Is that a bug?

Also maybe instead of "No tracks to measure" it should say "No tracks have been measured".

carrotIndustries, 2020-06-27:
> "No tracks to measure" it should say "No tracks have been measured".

The measurement updates continuously while moving, so it's not a one-shot measure-and-done thing.

> I can't figure out what the difference between "Measure track" and "Measure all tracks"

Measure tracks only measures the length of the selected tracks. Measure all tracks measures the length of all tracks of the same net as the selected track.

Timmmm, 2020-06-27:
> The measurement updates continuously while moving, so it's not a one-shot measure-and-done thing.

Ok makes sense!

> Measure all tracks measures the length of all tracks of the same net as the selected track.

Ahh ok, maybe change them to "Measure tracks" and "Measure net tracks" respectively? Maybe you can think of better names...