horizon-eda/horizon #447: Issue: Junction not selectable in shadow other mode

bob-u, 2020-06-30:
Junction cannot be selected when in 'shadow other' mode.

carrotIndustries, 2020-06-30:
As written on IRC, this is caused by all non as-is modes enabling the work layer only mode in the selection filter to prevent objects on other layers from being selected. Perhaps introducing a new kind of selection filter mode for this use case is worth it such that only objects on the work layer and non-layer objects can be selected.

bob-u, 2020-07-01:
Is there an IRC archive for #horizon chanel?

carrotIndustries, 2020-07-07:
> I might be misunderstanding however, isn't junction a work layer object?

It depends. If a junction is only connected to one layer, it's on that layer. If it's connected to more than one layer, it's on the non-layer layer.

So what's the optimal way the work layer only modes should interact with the selection filter without introducing too much complexity?

 - Don't do anything at all (not so great, leaves too much selectable)
 - Force work layer only mode (things that should be selectable aren't)
 - Work layer only + keep non-layer items selectable (keeps junctions that span multiple layers selectable)

bob-u, 2020-07-08:
What is non-layer layer?  I don't think I understand the concept - is this basically everything that is not current layer?  What about parts that have copper on top and bottom but not in the inner layers.

Functionally, when in work layer mode,  changing physical properties of objects that have presence on other layers would cause trouble and should not be allowed.  Track's junction is not one of them and should be selectable.

Maybe, for simplicity sake, when multi-layer object is selected board should switch to as-is view?

carrotIndustries, 2020-07-08:
> What is non-layer layer? I don't think I understand the concept - is this basically everything that is not current layer? What about parts that have copper on top and bottom but not in the inner layers.

https://github.com/horizon-eda/horizon/blob/master/src/canvas/selectables.hpp#L37 For objects on a particular layer (such as tracks and polygons), `layer` is set to the object the layer is on. For objects that aren't on any particular layer such as dimensions or junctions that connect multiple layers, `layer` is set to `10000`.

> What about parts that have copper on top and bottom but not in the inner layers.

Packages are always on the top/bottom package layer

> Maybe, for simplicity sake, when multi-layer object is selected board should switch to as-is view?

No, that would be confusing.

carrotIndustries, 2020-08-24:
Having though about it a bit more, the proper way would be to replace the single layer in selectables and targets with something similar to https://github.com/horizon-eda/horizon/blob/master/3rd_party/router/router/pns_layerset.h#L32. Vias and their  would then span all copper layers and be selectable if at least one copper layer is active.

anuejn, 2020-08-25:
sounds nice :)

carrotIndustries, 2020-08-26:
See the linked branch for the current work in progress state and check if I missed something. That branch includes some addition debugging info by showing the layers a junction spans in the HUD.

anuejn, 2020-08-27:
works fine for me when i drag the via :).
However, when I select the junction (via hover select) i cant drag the via (which might be another issue (because it also appears in "as is display mode"

bob-u, 2020-08-27:
Moving juctions works now.  I am curious if there is a reason for not enabling package selection in 'Shadow other' mode?

carrotIndustries, 2020-08-27:
> However, when I select the junction (via hover select) i cant drag the via

For me, it starts routing if "drag to start tracks" is enabled. If it's not, dragging moves the junctions.

>  I am curious if there is a reason for not enabling package selection in 'Shadow other' mode?

As far as the selection filter is concerned, there's only the work layer only setting that gets set by the layer mode.

Packages are a bit tricky. Currently, packages are on the top/bottom package layer and are selectable eve if the layer they're on isn't visible. This means that they're not selectable in work layer only mode unless the work layer is top/bottom package. I tried putting them on the layer range top copper - top courtyard, but this leaves them selectable if the work layer is set to silkscreen.

In the latest commit on the layer-range branch, packages are placed on the top/bottom copper layer, keeping them selectable if in work layer only mode on a copper layer.

closed 2020-08-29 by carrotIndustries

anuejn, 2020-08-30:
Thanks a lot :)

commit ed7d975a (2020-08-29): use layer ranges for vias&junctions

fixes #447