horizon-eda/horizon #448: Issue: cursor vs mouse pointer

bob-u, 2020-07-01:
As I was taking screenshots to illustrate selection list I noticed that when cursor snaps to an object near the mouse pointer and LMB is pressed coordinates are taken from the mouse pointer location instead of cursor location.  I would expect that Cursor location would take precedence?

[image]

bob-u, 2020-07-01:
`bool ImpBase::handle_click_release(GdkEventButton *button_event)
{
    if (button_event->button == 3) {canvas->set_selection_mode(CanvasGL::SelectionMode::HOVER);} // revert to hover select with left click
I think I like the behavior where RMB reverts back to hover mode, below.  For this to work perfectly hover selection would have to look different from actual selection.

    if (core->tool_is_active() && button_event->button != 2 && !(button_event->state & Gdk::SHIFT_MASK)) {
        ToolArgs args;
        args.type = ToolEventType::CLICK_RELEASE;
        args.coords = canvas->get_cursor_pos();
        args.button = button_event->button;
        args.target = canvas->get_current_target();
        args.work_layer = canvas->property_work_layer();
        ToolResponse r = core->tool_update(args);
        tool_process(r);
    }
    return false;
}`

carrotIndustries, 2020-07-01:
> I do see how getting mouse pointer position is a better option. 

Yes, snapping box selection to the grid is really pointless and annoying. (looking at you, kicad schematic editor)

> what is the purpose of the cursor when no tool is selected?

Some tools such as move use the cursor location as a starting point since snapping to things before doing the move is really useful in some cases.

bob-u, 2020-08-02:
Moved to discourse

closed 2020-08-02 by bob-u