horizon-eda/horizon #449: Junction in via

bob-u, 2020-07-02:
What is the purpose of junction in via?  Why is it selectable, can be duplicated, moved and copied apart from via itself?

carrotIndustries, 2020-07-02:
Because that's how it works internally: Tracks start/end at junctions (or pads) and vias are attached to junctions. Junctions are regular objects such as tracks and vias.

bob-u, 2020-07-02:
If there there's no valid task to be done with via's junction, then shouldn't selecting via's junction select via or via and junction (i.e. there's no point in selecting junction alone)?

bob-u, 2020-07-02:
I see that important tool operations selected for junction are applied to via+junction and only few menu items are available but do not do anything.  I am closing this ticket as this is quite minor issue in the scheme of things.

Perhaps one more improvement I could suggest would be for HUD to display via information when junction is selected, after-all no one want's to know more information or do operations on junction alone.

closed 2020-07-02 by bob-u