horizon-eda/horizon #450: Better keyboard control

bob-u, 2020-07-05:
This program has so many hidden features (at least to me) - using hover I can move objects with cursor keys w/o clicking or going through menus, and w/o being affected by cursor snap.  It works well and it offers precise x/y moves (fine step is also available).  What I would like to see is:

- when key auto-repeat starts, object location should update every step - right now position changes but object will not be redrawn until one lets go of the key.

- when track is highlighted - G mode should be the default move for keyboard move control.  To adjust track size one can still manipulate track's junction.

carrotIndustries, 2020-07-06:
> when key auto-repeat starts, object location should update every step - right now position changes but object will not be redrawn until one lets go of the key.

It does on my system.

> when track is highlighted - G mode should be the default move for keyboard move control. To adjust track size one can still manipulate track's junction.

You're referring to the interactive drag that uses the kicad router? That one doesn't support keyboard control.

bob-u, 2020-07-07:
> It does on my system.

I did some more testing and here what I see:
when pressing and holding an arrow key: object moves one step, then disappears and reappears after 12 steps, then disappears and reappears after 6 steps, then continues to reaper every 6 steps.  Step is equal grid size.

||/ Name           Version      Architecture Description
+++-==============-============-============-=================================
ii  libgtk-3-0:amd 3.22.30-1ubu amd64        GTK+ graphical user interface lib
ii  libgtk2.0-0:am 2.24.32-1ubu amd64        GTK+ graphical user interface lib

> You're referring to the interactive drag that uses the kicad router? That one doesn't support keyboard control.

That's probably what it is "drag and keep slope" can be only adjusted with mouse.  However it does not push and shove.

carrotIndustries, 2020-07-07:
> I did some more testing and here what I see:

Can you share a screen recording of what it looks like. Rather than gtk, often times, OpenGL drivers are a bit flakey.

bob-u, 2020-07-07:
> Can you share a screen recording of what it looks like. Rather than gtk, often times, OpenGL drivers are a bit flakey.

I cannot reproduce it at the moment.  I did install screen capture and in the process run apt update so it may have been the fix.  I will reply if the issue comes back.

bob-u, 2020-08-02:
Unable to reproduce

closed 2020-08-02 by bob-u