horizon-eda/horizon #453: Routing tracks crashes board editor

atoav, 2020-07-09:
## Problem

When I route tracks with any build after e04d8f33 the board editor crashes as soon as I try to draw a track from one end of the airwire to another. This happens with completely "fresh" projects without anything in it and regardless of the kind of parts used. 

The board editor will then exit with Status 134 and the console will show the following output:
```
rebuild took 0.000876
render took 319.795
render took 664.894
click select
tool delete
needs save 16912 1
rebuild took 0.002099
render took 634.518
tool route track
render took 707.714
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
```

This sounds very much like something tries to access something that doesn't exist and boom. Given the very obvious nature of the problem I guess that somebody else might have noticed already if this was a universal problem. Maybe it is specific to my system? It looks like this:
```
Operating System: Kubuntu 20.04
KDE Plasma Version: 5.18.5
KDE Frameworks Version: 5.68.0
Qt Version: 5.12.8
Kernel Version: 5.4.0-40-generic
OS Type: 64-bit
Processors: 16 × Intel® Core™ i9-9900K CPU @ 3.60GHz
Memory: 62,8 GiB of RAM
```

bob-u, 2020-07-09:
Here's a datapoint: Board editor continues to freeze my AMD Ryzen 2200G (APU system, which might be related to Mesa/GPU driver.  Otherwise, none of a  recent builds have resulted in error you describe on my Intel based system.

[image]

atoav, 2020-07-09:
I have a Nvidia RTX2080M, and don't have the same thing happening on Windows as well

carrotIndustries, 2020-07-09:
> Board editor continues to freeze my AMD Ryzen 2200G (APU Ubuntu (same system with Windows is fine), which might be related to Mesa/GPU driver. 

This is completely unrelated.

> maybe it is specific to my system? It looks like this:

Being on linux makes things easy/possible to debug on your end as I can't reproduce this:

 - Open the board editor
 - Find out its process id (easiest way is to use xprop and look for `_NET_WM_PID`)
 - Attach gdb with `sudo gdb -p <pid of board editor>`
 - Wait for gdb to load, enter `catch throw` to catch exceptions and `cont` to resume the board editor
 - Now do what causes it to crash
 - Gdb should return to the prompt
 - Enter `bt` in gdb to get a backtrace and post the result.

atoav, 2020-07-09:
Okay the gdb backtrace looks like this:
```
[New Thread 0x7f92e178f700 (LWP 7271)]

Thread 1 "horizon-imp" hit Catchpoint 1 (exception thrown), 0x00007f92eaabb762 in __cxa_throw ()
   from /lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007f92eaabb762 in __cxa_throw () at /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007f92eaab23be in std::__throw_out_of_range(char const*) () at /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x000055f0c5ea5b20 in std::__detail::_Map_base<horizon::ObjectRef, std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > >, std::allocator<std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > > >, std::__detail::_Select1st, std::equal_to<horizon::ObjectRef>, std::hash<horizon::ObjectRef>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::at(horizon::ObjectRef const&) (__k=..., this=<optimized out>)
    at /usr/include/c++/9/bits/stl_tree.h:1006
#3  std::unordered_map<horizon::ObjectRef, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > >, std::hash<horizon::ObjectRef>, std::equal_to<horizon::ObjectRef>, std::allocator<std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > > > >::at(horizon::ObjectRef const&) (__k=..., this=0x55f0c858f058)
    at /usr/include/c++/9/bits/unordered_map.h:1002
#4  horizon::Canvas::remove_obj(horizon::ObjectRef const&) (this=0x55f0c858f000, r=...) at src/canvas/canvas.cpp:78
#5  0x000055f0c5e0523d in PNS::PNS_HORIZON_IFACE::EraseView() (this=0x55f0ca1f6b20) at src/router/pns_horizon_iface.cpp:794
#6  0x000055f0c5d7f648 in PNS::ROUTER::movePlacing(VECTOR2<int> const&, PNS::ITEM*) (this=0x55f0ca0e2bd0, aP=..., aEndItem=0x0)
    at 3rd_party/router/router/pns_router.cpp:333
#7  0x000055f0c5d7f776 in PNS::ROUTER::Move(VECTOR2<int> const&, PNS::ITEM*) (this=0x55f0ca0e2bd0, aP=..., endItem=<optimized out>)
    at 3rd_party/router/router/pns_router.cpp:239
#8  0x000055f0c5e1333a in horizon::ToolRouteTrackInteractive::update(horizon::ToolArgs const&) (this=0x55f0ca173c10, args=...)
    at src/core/tools/tool_route_track_interactive.cpp:809
#9  0x000055f0c5ef26e8 in horizon::Core::tool_update(horizon::ToolArgs const&) (this=0x55f0c7edb730, args=...)
    at /usr/include/c++/9/bits/unique_ptr.h:360
#10 0x000055f0c5e35e67 in horizon::ImpBase::handle_cursor_move(horizon::Coord<long> const&) (this=0x55f0c7edb200, pos=...)
    at src/imp/imp.cpp:1495
#11 0x000055f0c5eadc98 in sigc::internal::signal_emit1<void, horizon::Coord<long> const&, sigc::nil>::emit(sigc::internal::signal_impl*, horizon::Coord<long> const&) (_A_a1=..., impl=0x55f0c8fedca0) at /usr/include/sigc++-2.0/sigc++/functors/slot_base.h:335
#12 sigc::signal1<void, horizon::Coord<long> const&, sigc::nil>::emit(horizon::Coord<long> const&) const
    (this=0x55f0c858f9a0, _A_a1=...) at /usr/include/sigc++-2.0/sigc++/signal.h:2955
--Type <RET> for more, q to quit, c to continue without paging--c
#13 horizon::CanvasGL::cursor_move(_GdkEvent*) (motion_event=0x55f0ca16cc80, this=0x55f0c858f000) at src/canvas/canvas_gl.cpp:335
#14 horizon::CanvasGL::on_motion_notify_event(_GdkEventMotion*) (this=0x55f0c858f000, motion_event=0x55f0ca16cc80) at src/canvas/canvas_gl.cpp:274
#15 0x00007f92ec991e89 in Gtk::Widget_Class::motion_notify_event_callback(_GtkWidget*, _GdkEventMotion*) () at /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#16 0x00007f92ebe914fb in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#17 0x00007f92eb024802 in g_closure_invoke () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007f92eb037f96 in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007f92eb04345d in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007f92eb0440d3 in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007f92ebe3bc23 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#22 0x00007f92ebcf7128 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#23 0x00007f92ebcf93db in gtk_main_do_event () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007f92eb9dff79 in  () at /lib/x86_64-linux-gnu/libgdk-3.so.0
#25 0x00007f92eba13106 in  () at /lib/x86_64-linux-gnu/libgdk-3.so.0
#26 0x00007f92eaf38fbd in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007f92eaf39240 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007f92eaf392e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007f92eb152fd5 in g_application_run () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#30 0x00007f92ec8c9502 in Gtk::Application::run(Gtk::Window&) () at /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#31 0x000055f0c5e48921 in horizon::ImpBase::run(int, char**) (this=0x55f0c7edb200, argc=<optimized out>, argv=<optimized out>) at src/imp/imp.cpp:889
#32 0x000055f0c5bb54b6 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at src/imp/imp_main.cpp:144
```

carrotIndustries, 2020-07-09:
Try this branch: https://github.com/horizon-eda/horizon/tree/router-crash-debug and post the output

pipatron, 2020-08-02:
I too was hit by this bug, first time I ever try horizon! Since OP didn't get back, I thought maybe I could help.

The backtrace from the crash on my first build, 7dfeadb231f382db9f96f3e6b22157831750733d (one commit after v1.2.0)

````
#0  0x00007f8790f0ebbe in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007f8790f0610a in std::__throw_out_of_range(char const*) () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x0000560f85541450 in std::__detail::_Map_base<horizon::ObjectRef, std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > >, std::allocator<std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > > >, std::__detail::_Select1st, std::equal_to<horizon::ObjectRef>, std::hash<horizon::ObjectRef>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::at(horizon::ObjectRef const&) (__k=..., this=0x560f88416658) at /usr/include/c++/9/bits/stl_tree.h:1006
#3  std::unordered_map<horizon::ObjectRef, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > >, std::hash<horizon::ObjectRef>, std::equal_to<horizon::ObjectRef>, std::allocator<std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > > > >::at(horizon::ObjectRef const&) (__k=..., this=0x560f88416658) at /usr/include/c++/9/bits/unordered_map.h:1002
#4  horizon::Canvas::remove_obj(horizon::ObjectRef const&) (this=0x560f88416600, r=...) at src/canvas/canvas.cpp:78
#5  0x0000560f854a1625 in PNS::PNS_HORIZON_IFACE::EraseView() (this=0x560f895c56e0) at src/router/pns_horizon_iface.cpp:801
#6  0x0000560f8541fdf4 in PNS::ROUTER::movePlacing(VECTOR2<int> const&, PNS::ITEM*) (this=0x560f895e3a10, aP=..., aEndItem=0x0) at 3rd_party/router/router/pns_router.cpp:333
#7  0x0000560f8541fef2 in PNS::ROUTER::Move(VECTOR2<int> const&, PNS::ITEM*) (this=0x560f895e3a10, aP=..., endItem=<optimized out>) at 3rd_party/router/router/pns_router.cpp:239
#8  0x0000560f854b289a in horizon::ToolRouteTrackInteractive::update(horizon::ToolArgs const&) (this=0x560f89773f20, args=...) at src/core/tools/tool_route_track_interactive.cpp:817
#9  0x0000560f8558a8af in horizon::Core::tool_update(horizon::ToolArgs const&) (this=0x560f87e78d68, args=...) at /usr/include/c++/9/bits/unique_ptr.h:360
#10 0x0000560f854d4d4e in horizon::ImpBase::handle_cursor_move(horizon::Coord<long> const&) (this=0x560f87e787c0, pos=...) at src/imp/imp.cpp:1542
#11 0x0000560f85548ec8 in sigc::internal::signal_emit1<void, horizon::Coord<long> const&, sigc::nil>::emit(sigc::internal::signal_impl*, horizon::Coord<long> const&) (_A_a1=..., impl=0x560f8841bf70) at /usr/include/sigc++-2.0/sigc++/functors/slot_base.h:335
#12 sigc::signal1<void, horizon::Coord<long> const&, sigc::nil>::emit(horizon::Coord<long> const&) const (this=0x560f88416fa0, _A_a1=...) at /usr/include/sigc++-2.0/sigc++/signal.h:2955
#13 horizon::CanvasGL::cursor_move(_GdkEvent*) (motion_event=0x560f897888c0, this=0x560f88416600) at src/canvas/canvas_gl.cpp:335
#14 horizon::CanvasGL::on_motion_notify_event(_GdkEventMotion*) (this=0x560f88416600, motion_event=0x560f897888c0) at src/canvas/canvas_gl.cpp:274
#15 0x00007f879323acf5 in Gtk::Widget_Class::motion_notify_event_callback(_GtkWidget*, _GdkEventMotion*) () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#16 0x00007f87927440fb in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#17 0x00007f879145efd2 in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007f8791471f06 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007f879147d078 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007f879147dedf in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007f87926eff12 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#22 0x00007f87925afc98 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#23 0x00007f87925b1e5b in gtk_main_do_event () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007f87922ab815 in  () at /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#25 0x00007f87922dda62 in  () at /usr/lib/x86_64-linux-gnu/libgdk-3.so.0
#26 0x00007f87913755fd in g_main_context_dispatch () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007f8791375880 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007f879137590f in g_main_context_iteration () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007f879158602d in g_application_run () at /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#30 0x00007f879318372e in Gtk::Application::run(Gtk::Window&) () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#31 0x0000560f854e7e66 in horizon::ImpBase::run(int, char**) (this=0x560f87e787c0, argc=<optimized out>, argv=<optimized out>) at src/imp/imp.cpp:890
#32 0x0000560f8526b712 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at src/imp/imp_main.cpp:144
````

[... cut ...]
src/schematic/schematic.cpp:395:22: warning: ‘*((void*)(& it)+16).horizon::LineNet::from’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  395 |                 auto from = line->from;
      |                      ^~~~
src/schematic/schematic.cpp:396:22: warning: ‘*((void*)(& it)+16).horizon::LineNet::to’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  396 |                 auto to = line->to;
      |                      ^~
````

Hope any of this helps, I have no problem testing other patches. I'd love to continue past the routing stage and make a complete board with horizon!

carrotIndustries, 2020-08-02:
> I suppose it has to do with some library or compiler version.

That's my take as well.

https://github.com/horizon-eda/horizon/blob/fea63e62432d8b2d9596de65f3e60915cfde8564/src/canvas/canvas.cpp#L87-L90

Mapping this to your output, I still don't see why it crashes since the object is present in the map as indicated by `ref count 1`. Perhaps you could add another debug print after line 90 to see if it really crashes there.

You might also try compiling with address / ub sanitizer enabled to see if these catch anything: https://gcc.gnu.org/onlinedocs/gcc/Instrumentation-Options.html

endofexclusive, 2020-08-03:
I believe I hit this one as well, on 14de3200d7e3e43df879dfd05cb4cfa2eae11d27.

Additional observation:
It was not possible to reproduce the bug when Horizon was compiled with `-O0` instead of `-O3`. 

This is how to trig it here:
- Hit `x` key to draw track
- Click a pad.
- Move mouse pointer a few pixels.

[image]

The program does not exit, but the draw track tool is aborted.

Attaching GDB:
```
#0  0x00007fc5dc619afd in __cxa_throw () at /lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x00007fc5dc61586b in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#2  0x0000563f8a9bb749 in std::__detail::_Map_base<horizon::ObjectRef, std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > >, std::allocator<std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > > >, std::__detail::_Select1st, std::equal_to<horizon::ObjectRef>, std::hash<horizon::ObjectRef>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<true, false, true>, true>::at(horizon::ObjectRef const&) (__k=..., this=0x563f8e633c58)
    at /usr/include/c++/8/bits/stl_tree.h:976
#3  0x0000563f8a9bb749 in std::unordered_map<horizon::ObjectRef, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > >, std::hash<horizon::ObjectRef>, std::equal_to<horizon::ObjectRef>, std::allocator<std::pair<horizon::ObjectRef const, std::map<int, std::pair<unsigned long, unsigned long>, std::less<int>, std::allocator<std::pair<int const, std::pair<unsigned long, unsigned long> > > > > > >::at(horizon::ObjectRef const&)
    (__k=..., this=0x563f8e633c58) at /usr/include/c++/8/bits/unordered_map.h:991
#4  0x0000563f8a9bb749 in horizon::Canvas::remove_obj(horizon::ObjectRef const&) (this=0x563f8e633c00, r=...)
    at src/canvas/canvas.cpp:78
#5  0x0000563f8a924c55 in PNS::PNS_HORIZON_IFACE::EraseView() (this=0x563f8e635860) at src/router/pns_horizon_iface.cpp:801
#6  0x0000563f8a89b671 in PNS::ROUTER::movePlacing(VECTOR2<int> const&, PNS::ITEM*) (this=0x563f90d652d0, aP=..., aEndItem=0x0)
    at 3rd_party/router/router/pns_router.cpp:333
#7  0x0000563f8a89b781 in PNS::ROUTER::Move(VECTOR2<int> const&, PNS::ITEM*) (this=0x563f90d652d0, aP=..., endItem=<optimized out>)
    at 3rd_party/router/router/pns_router.cpp:239
#8  0x0000563f8a9349ca in horizon::ToolRouteTrackInteractive::update(horizon::ToolArgs const&) (this=0x563f91d20a50, args=...)
    at src/core/tools/tool_route_track_interactive.cpp:817
#9  0x0000563f8aa0840e in horizon::Core::tool_update(horizon::ToolArgs const&) (this=0x563f8c36a9a8, args=...)
    at /usr/include/c++/8/bits/unique_ptr.h:342
#10 0x0000563f8a952d10 in horizon::ImpBase::handle_cursor_move(horizon::Coord<long> const&) (this=0x563f8c36a400, pos=...)
    at src/imp/imp.cpp:1542
#11 0x0000563f8a9c133d in sigc::internal::signal_emit1<void, horizon::Coord<long> const&, sigc::nil>::emit(sigc::internal::signal_impl*, horizon::Coord<long> const&) (_A_a1=..., impl=0x563f8d8752a0) at /usr/include/sigc++-2.0/sigc++/functors/slot_base.h:335
#12 0x0000563f8a9c133d in sigc::signal1<void, horizon::Coord<long> const&, sigc::nil>::emit(horizon::Coord<long> const&) const
    (this=<optimized out>, _A_a1=...) at /usr/include/sigc++-2.0/sigc++/signal.h:2955
#13 0x0000563f8a9c133d in horizon::CanvasGL::cursor_move(_GdkEvent*) (this=<optimized out>, motion_event=<optimized out>)
    at src/canvas/canvas_gl.cpp:335
#14 0x0000563f8a9c1c14 in horizon::CanvasGL::on_motion_notify_event(_GdkEventMotion*)
    (this=0x563f8e633c00, motion_event=0x563f8e17c000) at src/canvas/canvas_gl.cpp:274
#15 0x00007fc5de874c64 in Gtk::Widget_Class::motion_notify_event_callback(_GtkWidget*, _GdkEventMotion*) ()
    at /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#16 0x00007fc5dde4117b in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#17 0x00007fc5dcc4cc8d in g_closure_invoke () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007fc5dcc5fbbd in  () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007fc5dcc689ab in g_signal_emit_valist () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007fc5dcc6997f in g_signal_emit () at /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007fc5dddef324 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#22 0x00007fc5ddcaf976 in  () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#23 0x00007fc5ddcb1a83 in gtk_main_do_event () at /lib/x86_64-linux-gnu/libgtk-3.so.0
#24 0x00007fc5dd9b1465 in  () at /lib/x86_64-linux-gnu/libgdk-3.so.0
#25 0x00007fc5dd9e2112 in  () at /lib/x86_64-linux-gnu/libgdk-3.so.0
#26 0x00007fc5dcb6af2e in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#27 0x00007fc5dcb6b1c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#28 0x00007fc5dcb6b25c in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
#29 0x00007fc5dcd85a2d in g_application_run () at /lib/x86_64-linux-gnu/libgio-2.0.so.0
#30 0x00007fc5de7be645 in Gtk::Application::run(Gtk::Window&) () at /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#31 0x0000563f8a966a02 in horizon::ImpBase::run(int, char**)
    (this=this@entry=0x563f8c36a400, argc=<optimized out>, argv=<optimized out>) at src/imp/imp.cpp:890
#32 0x0000563f8a6e6ded in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at src/imp/imp_main.cpp:144
```

```
$ uname -a
Linux cl 4.19.0-9-amd64 #1 SMP Debian 4.19.118-2+deb10u1 (2020-06-07) x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 10 (buster)
Release:	10
Codename:	buster
$ c++ --version|head -n 1
c++ (Debian 8.3.0-6) 8.3.0
```

bob-u, 2020-08-03:
I am not a programmer but shouldn't line 94 be       

`` return !via != !track; // xor

pipatron, 2020-08-03:
Ok, I've done some more research, without digging too deep into the sources.

### The flatpak version shows the same issue
The more interesting is that this crash happens with the flatpak v1.2.0 installed according to the [instructions](https://flathub.org/apps/details/org.horizon_eda.HorizonEDA).

This kinda rules out my unique build setup I think, although I'm not terribly familiar with the flatpak system.

### -fsanitize=undefined / -fsanitize=address
I tried building the latest version with `-fsanitize=…` and I figured that this worked: `make CFLAGS="-fsanitize=address -fsanitize=undefined" LDFLAGS="-fsanitize=address -fsanitize=undefined"`

At first it didn't even start:
````
pipe@spacelab:~/src/horizon$ build/horizon-eda
=================================================================
==115228==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x6060000c7400 in thread T0:
  object passed to delete has wrong type:
  size of the allocated type:   64 bytes;
  size of the deallocated type: 48 bytes.
    #0 0x7f37f62d3625 in operator delete(void*, unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10a625)
    #1 0x55a382573774 in sigc::slot0<void>::~slot0() /usr/include/sigc++-2.0/sigc++/functors/slot.h:515
    #2 0x55a382573774 in sigc::slot<void, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::~slot() /usr/include/sigc++-2.0/sigc++/functors/slot.h:1761
    #3 0x55a382573774 in horizon::install_signal_exception_handler() src/util/exception_util.cpp:41
    #4 0x55a3821eb150 in main src/pool-prj-mgr/pool-prj-mgr-main.cpp:20
    #5 0x7f37f3999cc9 in __libc_start_main ../csu/libc-start.c:308
    #6 0x55a3821fd649 in _start (/home/pipe/src/horizon/build/horizon-eda+0x202649)

0x6060000c7400 is located 0 bytes inside of 64-byte region [0x6060000c7400,0x6060000c7440)
allocated by thread T0 here:
    #0 0x7f37f62d219f in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10919f)
    #1 0x55a3825736f0 in sigc::slot0<void>::slot0<void (*)()>(void (* const&)()) /usr/include/sigc++-2.0/sigc++/functors/slot.h:546
    #2 0x55a3825736f0 in sigc::slot<void, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::slot<void (*)()>(void (* const&)()) /usr/include/sigc++-2.0/sigc++/functors/slot.h:1774
    #3 0x55a3825736f0 in horizon::install_signal_exception_handler() src/util/exception_util.cpp:41

SUMMARY: AddressSanitizer: new-delete-type-mismatch (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10a625) in operator delete(void*, unsigned long)
==115228==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
==115228==ABORTING
````

I then ran the tool with the recommended `ASAN_OPTIONS`. Everything started out fine, the crash happened as usual when I tried to route, but I didn't get any output from the sanitizer until I closed the PCB editor. Then I got a few leaks:
````
pipe@spacelab:~/src/horizon$ ASAN_OPTIONS=new_delete_type_mismatch=0 build/horizon-eda
spawn /home/pipe/src/horizon/build/horizon-imp -b /home/pipe/share/project/amiga/hardware/SCSI2SD/scsi2sd-adapter-scsikit/board.json /home/pipe/share/project/amiga/hardware/SCSI2SD/scsi2sd-adapter-scsikit/top_block.json /home/pipe/share/project/amiga/hardware/SCSI2SD/scsi2sd-adapter-scsikit/vias /home/pipe/share/project/amiga/hardware/SCSI2SD/scsi2sd-adapter-scsikit/pictures 
rebuild took 0.007382
render took 38.4778
render took 460.405

match 1 0
tool route track
render took 399.521
render took 376.648
click select
match 1 0
tool route track
render took 507.099
render took 465.549
delete panel Package

=================================================================
==115264==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 16384 byte(s) in 1 object(s) allocated from:
    #0 0x7fd88e542628 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x107628)
    #1 0x7fd88b2811b8 in g_malloc (/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x561b8)

Direct leak of 6912 byte(s) in 27 object(s) allocated from:
    #0 0x7fd88e542628 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x107628)
    #1 0x7fd88835d704  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x20704)

Direct leak of 6400 byte(s) in 10 object(s) allocated from:
    #0 0x7fd88e542a2e in __interceptor_realloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x107a2e)
[... 62 lines cut ...]
    #0 0x7f7e154900b5 in strdup (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x920b5)
    #1 0x7f7e126e9fa4 in FcValueSave (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x1ffa4)

Indirect leak of 5152 byte(s) in 161 object(s) allocated from:
    #0 0x7f7e1550580e in calloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10780e)
    #1 0x7f7e126ead48  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x20d48)

Indirect leak of 2400 byte(s) in 75 object(s) allocated from:
    #0 0x7f7e1550580e in calloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10780e)
    #1 0x7f7e126ea237  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x20237)

Indirect leak of 576 byte(s) in 18 object(s) allocated from:
    #0 0x7f7e1550580e in calloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x10780e)
    #1 0x7f7e126ea0d6  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x200d6)

Indirect leak of 144 byte(s) in 3 object(s) allocated from:
    #0 0x7f7e15505628 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.5+0x107628)
    #1 0x7f7e126e437d in FcLangSetCreate (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x1a37d)

SUMMARY: AddressSanitizer: 51071 byte(s) leaked in 1054 allocation(s).
````

To be honest, they don't _seem_ related. Font stuff. Is this a locale-related setting? Running with `LC_ALL=C …` didn't make a difference.

My next step is to look at the sources and see if I can add some more prints where this occurs, although I haven't done any C++ in years.

carrotIndustries, 2020-08-04:
> The more interesting is that this crash happens with the flatpak v1.2.0 

Can confirm, that's quite unfortunate. Definitely seems like a compiler/libc++ issue.

pipatron, 2020-08-04:
Latest findings: I just built it with clang 9 instead of gcc 9 now it works like a charm.

carrotIndustries, 2020-08-04:
Good news: I downgraded gcc to 9.3 and can reproduce the issue on arch linux.

carrotIndustries, 2020-08-04:
This is getting really silly: https://gist.github.com/carrotIndustries/947c1e0e663859de5f03c277463a24e5

With gcc 9.3, the test program outputs
```
18446744073692774400
0
```

This means that gcc omits the last iteration in https://gist.github.com/carrotIndustries/947c1e0e663859de5f03c277463a24e5#file-main-cpp-L11 when inside of a loop and compiling with `-O3`. I tried to figure out which optimisation enabled by `-O3` triggers this bug, but couldn't trigger the bug by any combination of optimisations. I'll probably open a bug in the gcc bugtracker.

I came up with a workaround: https://github.com/horizon-eda/horizon/tree/gcc-9.3-fix If we agree that this is an acceptable fix, I'll do a 1.2.1 release that includes this patch.

pipatron, 2020-08-04:
Awesome! I can report that the gcc-9.3-fix branch indeed solves the issue on my platform.

closed 2020-08-05 by carrotIndustries

carrotIndustries, 2020-08-06:
Here's the release: https://github.com/horizon-eda/horizon/releases/tag/v1.2.1

Thanks to everyone helping to debug this nasty bug!

commit de5faf23 (2020-08-05): UUID: avoid shifting by more than bit width by casting to size_t

uu[i] gets promoted to an int (32 bit) but it shifted by up to 56 bits
on 64 bit platforms which results in undefined behaviour

this caused gcc 9.3 to emit code that miscalculated the hash in loops

cast uu[i] to size_t (64 bit on 64 bit platforms) to avoid undefined
behaviour

fixes #453
commit 37882e0f (2020-10-29): UUID: avoid shifting by more than bit width by casting to size_t

uu[i] gets promoted to an int (32 bit) but it shifted by up to 56 bits
on 64 bit platforms which results in undefined behaviour

this caused gcc 9.3 to emit code that miscalculated the hash in loops

cast uu[i] to size_t (64 bit on 64 bit platforms) to avoid undefined
behaviour

fixes #453