horizon-eda/horizon #454: Zero length traces

bob-u, 2020-07-09:
1. How do I avoid zero length traces?
2. What is the proper way to fix zero length trace?

In example below part is routed when horizontal, then rotated 90deg which results in zero length trace.  When zero trace is deleted, note that right trace looses it's connection like it's not there though it is still associated with the same net as the airwire that appears.  Only deleting right trace and redrawing it makes the airwire go away:

[image]

carrotIndustries, 2020-07-09:
Everything here works as expected and won't change in the near future:

> How do I avoid zero length traces?

Don't let tracks shrink to zero length.

> What is the proper way to fix zero length trace?

Move things or delete and redraw it

closed 2020-07-09 by carrotIndustries

bob-u, 2020-07-09:
Hi.  The issue is that I did not create zero length track.  Horizon-EDA created it when I rotated the part.  I am sure you know that however asking a user to avoid that seems like impossible request :)

carrotIndustries, 2020-07-09:
>  Horizon-EDA created it when I rotated the part. 

Well, this is what you get in some cases if rotating a package causes its pad to land on the start of the track it's attached to. Automatically deleting the track and reconnecting the pad to the next track wouldn't be worth the effort.

bob-u, 2020-07-15:
I should have perhaps ask this first: why zero length track is actually a problem and is being flagged?