horizon-eda/horizon #455: Feature Request: Board outline wizard

atoav, 2020-07-09:
This is fairly self explanatory and rather than a concrete proposal it is meant to be a starting point for discussion. 

The idea is to create a set of wizards similar to those in the Package Editor that help with the creation of boards to reduce the setup time when starting a new project.

So things like:
- Rectangular PCB (maybe with options like mounting holes, corner rounding, Ground Plane etc)
- Circular PCB
- all kind of generators for standard pcb sizes, HP widths etc
- User Template files (e.g. create a project click on "Save as Template", then be able to use it as a starting point in the future)

Of course this could be just a board-editor thing, but then there would be no way of doing practical things like adding planes automatically etc.

bob-u, 2020-07-09:
One thing that would be helpful is dimension tool that binds to and follows objects and, even better, could be used to keep objects apart at given dimensional constrain (for mounting holes, headers, ports, etc.

With that said, EasyEDA will never be as good as dedicated CAD drawing program.  Therefore, I think import feature with ability to snap objects to certain board features (beside grid) could be time better spent.

phkahler, 2021-01-10:
> One thing that would be helpful is dimension tool that binds to and follows objects and, even better, could be used to keep objects apart at given dimensional constrain (for mounting holes, headers, ports, etc.

@bob-u  Have you done any drawing with SolveSpace? It handles 2D drawing with an excellent constraint solver as you describe. If that's what you want I see a couple options.

1) the Solver is GPL3, so could be integrated into Horizon - seems like a lot of work.

2) If we added IDF export from SolveSpace (board only) and import in Horizon, you could do overall board geometry in SolveSpace using those capabilities and bring that design over to Horizon. On a related note, it doesn't look like Horizon supports IDF export. The spec can be found here:
http://www2.me.rochester.edu/courses/ME204/nx_help/en_US/graphics/fileLibrary/nx/tdoc_pcb/idf_v3.pdf

carrotIndustries, 2021-01-10:
> On a related note, it doesn't look like Horizon supports IDF export. The spec can be found here:

Are you aware of any existing free software that supports idf import? I couldn't tell if freecad supports idf import or note. Writing against a spec with no way to test isn't something I like to do.

phkahler, 2021-01-10:
>> Are you aware of any existing free software that supports idf import?

@carrotIndustries  I don't know of any free E-CAD that supports import. KiCad does export, and SolveSpace now has "import" in the form of what we call "linking" so we can check fit in an assembly (board only). The IDF reading code is here:
https://github.com/solvespace/solvespace/blob/master/src/importidf.cpp

Lines 293-521 read the file and most of the stuff above that is to create SolveSpace entities from things in the file plus a couple helpers. It's GPLv3 if you'd like to use it as a starting point.

This came about in response to this user request:
https://github.com/solvespace/solvespace/issues/625

Some of the ick in the code is because all the different outlines share essentially the same definitions in the file, so I started with just outline and holes but started trying to read an arc or line in one piece of code but handle all variants: board outline, keepout top, keepout bottom, in one place. I tried adding keepouts and it worked but that can add a huge number of entites and it slows us down (for now) so i commented it out.

The "standard" identifies weather a feature is owned by the E-Cad or M-Cad system and reads like it was meant to go back and forth during design, but I don't think that's really feasible. Make a board and bring it to M-Cad, or as I suggested make a board outline in M-Cad and use it as a basis for a PCB in E-Cad, but never try to round-trip the same file, there's not enough information to tie something back to internal data structures. I may add part outline/extrusions if people really want that in SolveSpace.

I am interested in increasing inter-op between SolveSpace and the free software EDA tools where it makes sense. One of our guys recently got colors to export properly in STEP files, so Horizon and KiCAD can both use it to make 3D part models now.

carrotIndustries, 2021-01-12:
> One of our guys recently got colors to export properly in STEP files, so Horizon and KiCAD can both use it to make 3D part models now.

From my perspective, the last missing piece for useful MCAD/ECAD integration is getting part models imported into solvespace, as mentioned in https://github.com/solvespace/solvespace/issues/88 

> I may add part outline/extrusions if people really want that in SolveSpace.

Depending on the use case, this could be an acceptable workaround for part models

phkahler, 2021-01-14:
> > One of our guys recently got colors to export properly in STEP files, so Horizon and KiCAD can both use it to make 3D part models now.
> 
> From my perspective, the last missing piece for useful MCAD/ECAD integration is getting part models imported into solvespace, as mentioned in [solvespace/solvespace#88](https://github.com/solvespace/solvespace/issues/88)

I can handle creating the b-rep stuff in SolveSpace but we need code to parse STEP files, which is not trivial.

> > I may add part outline/extrusions if people really want that in SolveSpace.
> 
> Depending on the use case, this could be an acceptable workaround for part models

I found some IDF (.emn) files to test with, but couldn't find any with parts. If someone can export a simple project from KiCAD with a few parts on both sides that would be enough to test with when I get back to it.