horizon-eda/horizon #456: Barier to entry

bob-u, 2020-07-09:
HorizonEDA is a well designed and good looking EDA with advanced features such as auto router and parts library, and added recently, colored tracks.

Still, there are many improvements that can be made.  Amazingly features are added every week. 

Given that decision had been made to release 1.0 I presume the intention is to cater to wider audience.  Therefore I think it is worthwhile to identify the main barriers to entry:

1. Key bindings and availabke tools.  It had been a focus to make the features accessible and Space menu had addressed this directly with great result, IMO.  Global key bindings are being considered and will be a nice improvement.  It was also suggested to improve search and, although Lukas has shut it down I think adding search facility to preferences would make the program more approachable for beginners.

2. Another barier to entry is one of HorizonEDA main ‘selling point’ - part library.  Part editor has many features and considerable effort had been made to make it powerful but I would argue that it is not beginner friendly.  To give an example, part library introduces new vocabulary and for the beginner it is not intuitive to figure out what needs to be done first.  Pool manager offers many options but does not hold hands or guide enough for beginner not just to create parts but also understand what is happening and how to reuse or build upon objects created.

3. There is also some unexpected behavior that I will post in separate tickets on, such as zero length tracks, for example. 

4. Another barier to entry is no way to import existing projects.  This can be handled multiple ways and probably best to discuss in separate ticket.

5. I would also like to propose to change the name to HorizonEDA for better google search results.

This is a well designed software with great features and intention of this post is to consider what could be done in order to appeal to wider audience.  Thanks for making it.

carrotIndustries, 2020-07-09:
> HorizonEDA is a well designed and good looking EDA 

Let's keep it that way.

> It was also suggested to improve search and, although Lukas has shut it down for now I think adding search facility to preferences would make the program more approachable for beginners.

No, it won't. People can explore everything that the preferences window has to offer in less than 3 min by clicking around. If your intent was to use this as an enabler for more preferences, you'll be disappointed as I usually try to reduce the complexity in the preferences dialog. More preferences will lead to more people finding odd things like [#452] and that's not really what I want.

> Pool manager offers many options but does not hold hands or guide enough for beginner not just to create parts but also understand what is happening and how to reuse or build upon objects created.

Any suggestions apart from a link to the docs?

> Another barrier to entry is no way to import existing projects.

Writing a useful importer that doesn't suck is a huge effort that I don't really see worthwhile. I prefer to spend my time on improving things for people that use Horizon EDA for cool real-world projects such as https://github.com/apertus-open-source-cinema/axiom-micro-mainboard/tree/r3 So don't expect me to write a KiCad/Eagle/whatever importer in the next years.

> I would also like to propose to change the name to HorizonEDA for better google search results.

Nope, I don't like the weird camel case and it probably won't make a difference. Googling "Horizon EDA" or "Horizon PCB", yields a first page full of relevant results. (for me at least)

Vorlent, 2020-07-31:
>Any suggestions apart from a link to the docs?

The docs aren't good enough right now. I tried to make a part and although the software itself was easy to use I was constantly wondering what the options do and in what order I have to go through the steps. Figuring that out by myself with just the documentation was more trouble than it should have been.

It's one thing to create an insufficiently documented and badly designed part for myself but if I wanted to upload my part to the pool it should follow some basic rules and be of a reasonably high quality. The documentation mentions a review process but if the rules aren't clear I'm just going to waste the time of the reviewers because I have no idea what they consider to be a passable part.

As I am writing this I noticed the existence of the part wizard which is basically the solution to my first paragraph. I guess it just needs to be a bit more prominent because I simply didn't know it existed.

A solution to the second paragraph would be more documentation in the pool itself. For example the entity editor has text fields.
Name, Manufacturer, Prefix. A README.MD file in symbols/ could explain what should be put into these fields to pass the review. The manufacturer name might be obvious but there could be multiple ways to spell the same manufacturer. The pool documentation would then point out that you have to check if the manufacturer already exists and that you must use the existing name if it does. I have no idea how prefixes are chosen so I can't even fill that field otherwise I'd get it wrong 100% of the time.

Vorlent, 2020-07-31:
It may not be obvious but I have almost zero experience with electronics. I'm primarily a software developer trying this out as a hobby.

carrotIndustries, 2020-07-31:
> As I am writing this I noticed the existence of the part wizard which is basically the solution to my first paragraph. I guess it just needs to be a bit more prominent because I simply didn't know it existed.

Any ideas how to make it more prominent? I though that a blue button was sufficient.

> The manufacturer name might be obvious but there could be multiple ways to spell the same manufacturer. 

That's why there's autocompletion.

> Name, Manufacturer, Prefix. A README.MD file in symbols/ could explain what should be put into these fields to pass the review. The manufacturer name might be obvious but there could be multiple ways to spell the same manufacturer.

We have the infrastructure for easily adding these help icons. It's just that someone has to come up with useful content and places to put them.
[image]

Vorlent, 2020-08-01:
Regarding the manufacturer selection I just used that as a simple example because I was lazy and didn't want to discuss 6 different text fields when I only wanted to get my idea across.

The problem with the parts wizard is that it's under the package tab. When I started using HorizonEDA I clicked on the "parts" tab where I could select an existing package. There was simply no need for me to go to the package tab and that's why I didn't see it. The UI is guiding you to create . Shouldn't the button be on the parts tab?

Help icons are okay with me but I can't help with that because as I said I simply don't know what the industry specific meaning is.

I'd like to thank you for adding tool icons to the left side of the symbol editor in version 1.2. The keybindings themselves are intuitive to use and I have no problem with that but adding icons makes it much easier to discover and remember the keybindings for a first time user like me.

Vorlent, 2020-08-01:
Also there is a problem with the "Part Wizard" button as is. If you don't select a package it will be greyed out and that also means it is not blue.

carrotIndustries, 2020-08-01:
> The problem with the parts wizard is that it's under the package tab.
> Shouldn't the "Parts Wizard" button be on the parts tab?

You're using an old version: It got moved to the parts tab a few months ago.

Vorlent, 2020-08-01:
I updated HorizonEDA before I wrote these two posts and even mentioned that I use the newest version in that comment. Yes, that problem is fixed but for some strange reason I still saw the old pool manager on 1.2. I can't explain why that happened but feel free to ignore the comments I wrote.

carrotIndustries, 2020-08-01:
> I can't explain why that happened

The pool/project manger is a single-instance application, so all windows belong to the same process. To get the new pool manager after updating, you'll first need to close all windows so that a new instance is launched.