horizon-eda/horizon #457: Board editor crashing when placing via using non-default pool

frmdstryr, 2020-07-10:
I have one project where whenever I try to place a via (by pressing pv then selecting the net) the board editor is crashing. If I try to place a via using the toggle during drawing a connection line the via shows up but when I click to place it, it just disappears (and does not crash for this).

It only happens in one project (it's using a custom pool that is inheriting the default pool, idk if that matters), the others that use the default pool all are working fine.

```c
Thread 1 "horizon-imp" received signal SIGSEGV, Segmentation fault.
0x0000555555914ac5 in horizon::Padstack::Padstack (this=0x555558028528, ps=...) at src/common/layer_provider.hpp:7
7       class LayerProvider {
(gdb) bt
#0  0x0000555555914ac5 in horizon::Padstack::Padstack(horizon::Padstack const&) (this=0x555558028528, ps=...) at src/common/layer_provider.hpp:7
#1  0x0000555555974142 in horizon::Via::Via(horizon::UUID const&, horizon::Padstack const*) (this=0x5555580284d0, uu=..., ps=<optimized out>) at src/util/uuid_ptr.hpp:30
#2  0x0000555555abf323 in std::pair<horizon::UUID const, horizon::Via>::pair<horizon::UUID&, 0ul, horizon::UUID&, horizon::Padstack const*&, 0ul, 1ul>(std::tuple<horizon::UUID&>&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&, std::_Index_tuple<0ul>, std::_Index_tuple<0ul, 1ul>) (__tuple2=..., __tuple1=..., this=0x5555580284c0) at /usr/include/c++/10/bits/move.h:76
#3  0x0000555555abf323 in std::pair<horizon::UUID const, horizon::Via>::pair<horizon::UUID&, horizon::UUID&, horizon::Padstack const*&>(std::piecewise_construct_t, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&>)
    (__second=..., __first=..., this=0x5555580284c0) at /usr/include/c++/10/tuple:1678
#4  0x0000555555abf323 in __gnu_cxx::new_allocator<std::_Rb_tree_node<std::pair<horizon::UUID const, horizon::Via> > >::construct<std::pair<horizon::UUID const, horizon::Via>, std::piecewise_construct_t const&, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&> >(std::pair<horizon::UUID const, horizon::Via>*, std::piecewise_construct_t const&, std::tuple<horizon::UUID&>&&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&&) (__p=0x5555580284c0, this=<optimized out>)
    at /usr/include/c++/10/ext/new_allocator.h:150
#5  0x0000555555abf323 in std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<horizon::UUID const, horizon::Via> > > >::construct<std::pair<horizon::UUID const, horizon::Via>, std::piecewise_construct_t const&, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&> >(std::allocator<std::_Rb_tree_node<std::pair<horizon::UUID const, horizon::Via> > >&, std::pair<horizon::UUID const, horizon::Via>*, std::piecewise_construct_t const&, std::tuple<horizon::UUID&>&&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&&) (__p=0x5555580284c0, __a=...) at /usr/include/c++/10/bits/alloc_traits.h:512
#6  0x0000555555abf323 in std::_Rb_tree<horizon::UUID, std::pair<horizon::UUID const, horizon::Via>, std::_Select1st<std::pair<horizon::UUID const, horizon::Via> >, std::less<horizon::UUID>, std::allocator<std::pair<horizon::UUID const, horizon::Via> > >::_M_construct_node<std::piecewise_construct_t const&, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&> >(std::_Rb_tree_node<std::pair<horizon::UUID const, horizon::Via> >*, std::piecewise_construct_t const&, std::tuple<horizon::UUID&>&&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&&) (this=0x555556a6c9a8, __node=0x5555580284a0) at /usr/include/c++/10/bits/stl_tree.h:618
#7  0x0000555555abf323 in std::_Rb_tree<horizon::UUID, std::pair<horizon::UUID const, horizon::Via>, std::_Select1st<std::pair<horizon::UUID const, horizon::Via> >, std::less<horizon::UUID>, std::allocator<std::pair<horizon::UUID const, horizon::Via> > >::_M_create_node<std::piecewise_construct_t const&, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&> >(std::piecewise_construct_t const&, std::tuple<horizon::UUID&>&&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&&) (this=0x555556a6c9a8)
    at /usr/include/c++/10/bits/stl_tree.h:635
#8  0x0000555555abf323 in std::_Rb_tree<horizon::UUID, std::pair<horizon::UUID const, horizon::Via>, std::_Select1st<std::pair<horizon::UUID const, horizon::Via> >, std::less<horizon::UUID>, std::allocator<std::pair<horizon::UUID const, horizon::Via> > >::_M_emplace_unique<std::piecewise_construct_t const&, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&> >(std::piecewise_construct_t const&, std::tuple<horizon::UUID&>&&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&&) (this=0x555556a6c9a8)
    at /usr/include/c++/10/bits/stl_tree.h:2414
#9  0x0000555555c0c388 in std::map<horizon::UUID, horizon::Via, std::less<horizon::UUID>, std::allocator<std::pair<horizon::UUID const, horizon::Via> > >::emplace<std::piecewise_construct_t const&, std::tuple<horizon::UUID&>, std::tuple<horizon::UUID&, horizon::Padstack const*&> >(std::piecewise_construct_t const&, std::tuple<horizon::UUID&>&&, std::tuple<horizon::UUID&, horizon::Padstack const*&>&&) (this=<optimized out>) at /usr/include/c++/10/bits/stl_map.h:576
#10 0x0000555555c0c388 in horizon::ToolPlaceVia::create_attached() (this=0x55555818ed60) at src/core/tools/tool_place_via.cpp:36
#11 0x0000555555bd34a5 in virtual thunk to horizon::ToolPlaceJunction::begin(horizon::ToolArgs const&) () at src/core/tools/tool_place_junction.hpp:35
#12 0x0000555555bcabdc in horizon::Core::tool_begin(horizon::ToolID, horizon::ToolArgs const&, horizon::ImpInterface*, bool) (this=<optimized out>, tool_id=<optimized out>, 
    tool_id@entry=horizon::ToolID::PLACE_VIA, args=..., imp=0x7fffffffc12c, transient=transient@entry=false) at /usr/include/c++/10/bits/unique_ptr.h:420
#13 0x0000555555af52f0 in horizon::ImpBase::tool_begin(horizon::ToolID, bool, std::set<horizon::SelectableRef, std::less<horizon::SelectableRef>, std::allocator<horizon::SelectableRef> > const&, std::unique_ptr<horizon::ToolData, std::default_delete<horizon::ToolData> >) (this=0x555556a6bf10, id=horizon::ToolID::PLACE_VIA, override_selection=<optimized out>, sel=std::set with 93825019772986 elements<error reading variable: Cannot access memory at address 0x18>, data=std::unique_ptr<horizon::ToolData> = {...})
    at /usr/include/c++/10/bits/unique_ptr.h:420
#14 0x0000555555af54ca in horizon::ImpBase::handle_tool_action(horizon::ActionConnection const&) (this=<optimized out>, conn=...) at /usr/include/c++/10/bits/stl_tree.h:211
#15 0x0000555555af7c3f in std::function<void (horizon::ActionConnection const&)>::operator()(horizon::ActionConnection const&) const (__args#0=..., this=0x7fffffffc460) at /usr/include/c++/10/bits/std_function.h:617
#16 0x0000555555af7c3f in horizon::ImpBase::trigger_action(std::pair<horizon::ActionID, horizon::ToolID> const&) (this=<optimized out>, action=...) at src/imp/imp.cpp:1090
#17 0x0000555555af880a in horizon::ImpBase::handle_action_key(_GdkEventKey*) (this=0x555556a6bf10, ev=<optimized out>) at /usr/include/c++/10/bits/stl_pair.h:266
#18 0x00007ffff46fe1cb in  () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#19 0x00007ffff2f847fb in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#20 0x00007ffff152b10d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007ffff153dbf1 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#22 0x00007ffff15460af in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#23 0x00007ffff154712f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#24 0x00007ffff30cc534 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#25 0x00007ffff30ecf0b in gtk_window_propagate_key_event () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#26 0x00007ffff30f08fb in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#27 0x00007ffff46fa897 in Gtk::Widget::on_key_press_event(_GdkEventKey*) () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#28 0x00007ffff46fcab4 in Gtk::Widget_Class::key_press_event_callback(_GtkWidget*, _GdkEventKey*) () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#29 0x00007ffff2f848f7 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#30 0x00007ffff152b346 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#31 0x00007ffff15463cd in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
[... cut ...]
#42 0x00007ffff463cd2a in Gtk::Application::run(Gtk::Window&) () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#43 0x0000555555b094b5 in horizon::ImpBase::run(int, char**) (this=this@entry=0x555556a6bf10, argc=<optimized out>, argv=<optimized out>) at /usr/include/glibmm-2.4/glibmm/refptr.h:259
#44 0x0000555555850356 in main(int, char**) (argc=<optimized out>, argv=<optimized out>) at src/imp/imp_main.cpp:144
```

The project is attached,  I can push the pool if needed.
[TB67S209.tar.gz](https://github.com/horizon-eda/horizon/files/4900187/TB67S209.tar.gz)

frmdstryr, 2020-07-10:
I changed the `ToolPlaceVia::create_attached` to split out the variables and the `get_via_padstack_uuid` is returning a uuid of all zero which is then causing `get_padstack` to return a nullptr.

frmdstryr, 2020-07-10:
Sorry for the noise  I figured it out.  

I don't know how it got like that but there were no rules under the Vias rules section, after adding a rule there it works.

bob-u, 2020-07-10:
This is still a bug, isn’t it?

frmdstryr, 2020-07-10:
I guess must have accidentally clicked the "-" button on the via rules tab. 

Ideally horizon should prevent saving with no via rules or force a default if none exists.

commit 77bd55ce (2020-07-10): tool place via: gracefully handle missing via rules

fixes #457
closed 2020-07-10 by carrotIndustries