horizon-eda/horizon #461: Unit editor - order isn't indicated

znfgnu, 2020-07-17:
Would be useful when rewriting a lot of data

carrotIndustries, 2020-07-17:
Can you elaborate? The pins in a unit don't have any natural order as they're stored in a `map` with their `uuid` as key. To make things more user-friendly, the pins get sorted alphabetically in the unit editor.

znfgnu, 2020-07-17:
My case was rewriting pins description from stm32f405 datasheet (https://www.st.com/resource/en/datasheet/dm00037051.pdf). Pins included in lqfp64 package were scattered abound the table, so when I found the next I was going to enter, I wanted to make sure what was the number of the last pin I entered. Even simple counter would do the job for me. Is my approach correct?

carrotIndustries, 2020-07-17:
The unit editor is far from the best tool for this job:
 - When adding new parts, use the part wizard as this one makes the pin-pad mapping way easier
 - For STM32 devices, use [this script](https://github.com/horizon-eda/horizon/blob/master/scripts/stm32_to_json.py) to extract the pin names from the STM32CubeMX application and import the generated json into the part wizard

carrotIndustries, 2020-12-29:
Closing due to inactivity. If any questions remain, feel free to ask on the forum.

closed 2020-12-29 by carrotIndustries