horizon-eda/horizon #462: Ease of use: Part Editor button

bob-u, 2020-07-18:
Part editor needs it's own button on the start screen.  Even if pool manager and part editor share the same window, it would be more intuitive for beginners to provide button for part editor.
[image]

[chip.zip](https://github.com/horizon-eda/horizon/files/4942335/chip.zip)

carrotIndustries, 2020-07-18:
Don't get it. The part editor itself doesn't to that much and is certainly more confusing without the context given by the pool manager.

bob-u, 2020-07-18:
1 pool manager is an alien concept at first, even users proficient with other EDA programs don't know what pool is for at first.  (Renaming pool manager to parts pool manager or part manager would also lessen the confusion).
2 everyone working on electronics knows what part is and that if part to be placed is not found - it needs to be created
3 however, how do you create a part if there is not button for it?
4 link to pool manager is not intuitive as everything else has a button, one would expect part manager or part editor to have one also.

Second best place to offer a button for part editor is within the part browser.

carrotIndustries, 2020-07-18:
Oh, looks like you're referring to the 'project manger' with 'start screen'? Please be more precise when referring to things or include explanatory screenshots.

Even if there was such a button, often times more than just the part editor is needed to create a new part.

> 4 link to pool manager is not intuitive as everything else has a button, one would expect part manager or part editor to have one also.

Nope, it's not related to the project.

As said elsewhere, the whole application is built around having at least skimmed over https://horizon-eda.readthedocs.io/en/latest/pool-why.html The pool is built the way it is to be flexible and accommodate many uses cases, not to be beginner-friendly. Yes, KiCad or other PCB design tools with simpler part structure might be easier to get started with, but will make life harder on the long run.

closed 2020-07-18 by carrotIndustries

bob-u, 2020-07-18:
> Oh, looks like you're referring to the 'project manger' with 'start screen'? Please be more precise when referring to things or include explanatory screenshots.

Project manager window does not have any name as far as I can see.   Might be be a good idea to add it to the window title.

> Nope, it's not related to the project.

 I have read that doc, prior however and I am familiar with it.  Because of new nomenclature this program introduces, I would suggest to include a link to the https://horizon-eda.readthedocs.io/en/latest/pool-why.html  within the project manager. 

I think you're misunderstanding the main point of this ticket.  Project manager window offers four large buttons for schematic, board, part browser, and pool cache, yet button to access the main feature of Horizon EDA is missing.  Browser link to the Pool Manager is not intuitive, especially with such large buttons near by.

carrotIndustries, 2020-07-18:
> Might be be a good idea to add it to the window title.

Don't see why that'd add any value. "Project name - application name" explains all that's needed.

> yet button to access the main feature of Horizon EDA is missing

The project manager isn't the main entry point to the application and shouldn't be cluttered with things unrelated to the current project. The start screen with the list of recent pools and project is.

bob-u, 2020-07-19:
Hi.  My opinions are from perspective of a new user.  Majority of users, I believe, learn by using the software and the more intuitive the interface is, the better user experience, and easier learning curve.

> Don't see why that'd add any value. "Project name - application name" explains all that's needed.

It would be easier to describe the window in tickets if it had a title.  If you recall I had difficulty describing which window I was talking about - above.

> The project manager isn't the main entry point to the application and shouldn't be cluttered with things unrelated to the current project. The start screen with the list of recent pools and project is.

Makes sense.  I would suggest to use Home button instead of '<' to better indicate the starting point:

[image]

Maybe this button would help me remember that if I close Pool Manager window, instead of clicking home button, I will exit the program.  Believe me - it happens a lot.

Now, once project had been created, clicking on change button next to pool should bring up a dialog to change the pool - all I'm getting is current pool that is already selected and nothing to add/remove/enable.  It seems to dialog that is accessible via settings->pools should be available here.

Finally I would suggest to add 'Open' or 'Manage' button next to pool, instead of a hot-link, to better convey to the user that this is they way to manage the pool.

EDIT: one more thing: In the Home window (Horizon EDA): given that select boxes (boxes with a list of pools and projects) do not have buttons next to them it is not obvious to click to open them.  It would be helpful to have  'Make Selection' above the list views to  nudge new user to make a choice by clicking.