horizon-eda/horizon #463: Feature request: Change filenames/paths

jbayless, 2020-07-19:
Hello,

I'm using Horizon 1.1.1 Blue sky commit 63ade81b on Windows. Having a wonderful time with it so far.

It seems like once I create a new padstack/package/part/etc, its file path cannot be changed. Sometimes I want to change a filepath, once I realize that eg. a package is manufacturer specific so I want to move it to the manufacturer's folder, or a unit is part of a series of functionally identical parts so I want to change the name to LT289x from LT2892 (to make up a random example).

There doesn't appear to be a "Save As" option in the editor, or any way to edit the object's path from the UI. I can change the object's name but not where its data is saved. I could just manually move or rename the file via the Windows file explorer, but I'm worried about whether that would result in breaking internal references within Horizon.

Do you think it would be possible and desirable to allow an object's file path to be updated through the Horizon UI?

carrotIndustries, 2020-07-19:
>  I could just manually move or rename the file via the Windows file explorer, but I'm worried about whether that would result in breaking internal references within Horizon.

There isn't much to worry about if adhering to these two rules:
 - Every object must be in in its respective subdirectory: Packages in packages and so on. The location in this subdirectory as well as the file name (as long as it ends in `.json`) don't matter as they're rediscovered during pool update.
 - For packages, always move the whole directory with the package.json in it
 - Do a pool update after moving around things to avoid surprises

> Do you think it would be possible and desirable to allow an object's file path to be updated through the Horizon UI?

Definitely something to consider. I'm thinking about a context menu item in the pool browsers named "Move/rename file".

jbayless, 2020-07-20:
Ah, thanks for the instructions. If Horizon can rediscover all the files, that's not so worrying then!

> I'm thinking about a context menu item in the pool browsers named "Move/rename file".

I think that would be the best way to implement this feature. Thank you!

commit 14de3200 (2020-08-02): pool manager: move/rename items

closes #463
closed 2020-08-02 by carrotIndustries